PhilipsHue/flutter_reactive_ble #231: Rename  any Connection to BleConnection

nicolasgarnet, 2021-02-25:
Hi,

This one might be unnecessary or too picky, but this library (as for my case) might sit next to a lot of other libraries mentionning a "connection state" such as Wi-fi, internet connectivity, http/mqtt clients, etc.

It might be more readable for library users to have any "Connection"mentionned renamed as BleConnection such as ConnectionStateUpdate class as a BleConnectionStateUpdate, DeviceConnectionState enum as DeviceBleConnectionState.

werediver, 2021-02-25:
Hi, @nicolasgarnet.

While I understand your concern, I think the real issue in such a situation wouldn't be naming, but rather mixing too many and/or unrelated concerns in a single file. When you have to, you can always use qualified imports either for the whole library or for certain types from it (equally applicable to flutter_reactive_ble and any other library.

```dart
import 'package:flutter_reactive_ble/flutter_reactive_ble.dart'; // Let's assume most of the types are named uniquely
import 'package:flutter_reactive_ble/flutter_reactive_ble.dart' as ble show ConnectionStateUpdate; // Let's assume `ConnectionStateUpdate` needs to be disambiguated

void f() {
  final flutterReactiveBle = FlutterReactiveBle(); // Refer the type as usual
  final ble.ConnectionStateUpdate someBleConnectionStateUpdate = null; // Refer the type with qualification `ble.`
  // ...
}
```

If this doesn't work, you may need to `hide` the clashing types in the first import, but I don't think this is needed.

Or you can just stick with a qualified import.

```dart
import 'package:flutter_reactive_ble/flutter_reactive_ble.dart' as ble;
```

werediver, 2021-02-28:
I am inclined to close this issue. I hope my comments on qualified and refined imports help to resolve any issue of this sort.

closed 2021-02-28 by werediver