PhilipsHue/flutter_reactive_ble #232: How to send data?

NikitaMasev, 2021-02-27:
How can i send data to the device? The connection to the device is established, but it is not clear what to do next to send data.

remonh87, 2021-02-27:
You can send data by using the writeCharacteristic operation. You probably need the writeCharacteristicWithResponse  operation.

NikitaMasev, 2021-02-27:
> You can send data by using the writeCharacteristic operation. You probably need the writeCharacteristicWithResponse operation.

Yes i saw. But first param of this function - QualifiedCharacteristic, which needs characteristicId, serviceId and deviceId. deviceId i can get from DiscoveredDevice after scanning devices. But how can i get characteristicId and serviceId ?

remonh87, 2021-02-27:
You can either use ‘discoverServices’ to see which services and characteristics the device has or you can check the docs of the peripheral you connect with

NikitaMasev, 2021-02-27:
> You can either use ‘discoverServices’ to see which services and characteristics the device has or you can check the docs of the peripheral you connect with

But why can't i just use deviceId? 
Early, i used flutter_bluetooth_serial and it require only deviceId.

werediver, 2021-02-27:
It's because "flutter_bluetooth_serial" implements a specific protocol (RFCOMM) on top of BLE. You can think of it as if the service ID and characteristic ID are hard-coded in that library (though, technically this is incorrect, because RFCOMM is implemented over L2CAP, a level of BLE stack at which services and characteristics do not exist [yet, GATT introduces them higher in the stack of protocols]).

"flutter_reactive_ble" only provides you an interface to GATT BLE protocol, which operates with services and characteristics. You can still implement any data transfer on top of it.

L2CAP support would be a nice addition to our library (this is being tracked in #191), but it's not going to happen any time soon.

closed 2021-03-06 by remonh87