PhilipsHue/flutter_reactive_ble #234: Wanted devices not fund but plenty of other random devices found.

patolax, 2021-02-28:
I just downloaded the repo and ran the example with the latest flutter_reactive_ble: ^2.7.3 version. The image below shows Bluetooth devices found on my PC, Android, IPhone and example flutter app running on android.

As you can see flutter app is not listing LAP-00 or my IPhone. But Android device scan is not showing my Iphone not sure why that is. However flutter app shows so many random devices not found by any of the other devices. 

Am I doing something wrong, why isnt the flutter app on android showing correct devices with the names? Is there away to resolve this? 

[image]

werediver, 2021-02-28:
If you would use the suggested issue template, one of the questions would be whether you have tried doing whatever you consider to be your issue in a generic BLE scanner app line "nRF Connect" and whether the result is different from what you are experiencing when using the library.

Because you disregarded the template, I have to ask you to go through this check in this message.

patolax, 2021-03-03:
Thanks that worked. I ran a GATT server on my PC and the sample app picked it up.

closed 2021-03-03 by patolax