PhilipsHue/flutter_reactive_ble #235: Exception: GenericFailure<WriteCharacteristicFailure

bobosette, 2021-03-01:
Hi. I have finally fixed the data transfer using subscribetocharacteristic.listen(), but I found a bug: now when I use flutter_reactive_ble to writecharacteristicwithresponse() on quite new android device (android 10.0 or newer) everything works fine, but if I use it on older version (android 7.0) I get this error:

_Exception (Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "GATT exception from MAC address B0:B4:48:CD:C7:84, with type BleGattOperation{description='CHARACTERISTIC_WRITE'}"))

How can I solve?

Thank you

remonh87, 2021-03-01:
@bobosette  normally Android also provides a statuscode along with it can you send the code or is this all that you received?

bobosette, 2021-03-01:
Unfortunately that's it!!! Nothing more 🤷🏻‍♂️

bobosette, 2021-03-04:
This is the StackTrace:

[ERROR:flutter/lib/ui/ui_dart_state.cc(186)] Unhandled Exception: Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "GATT exception from MAC address B0:B4:48:CD:C7:84, with type BleGattOperation{description='CHARACTERISTIC_WRITE'}")
E/flutter (12429): #0      Result.dematerialize.<anonymous closure> (package:flutter_reactive_ble/src/model/result.dart:22:13)
E/flutter (12429): #1      Result.iif (package:flutter_reactive_ble/src/model/result.dart:34:21)
E/flutter (12429): #2      Result.dematerialize (package:flutter_reactive_ble/src/model/result.dart:15:28)
E/flutter (12429): #3      ConnectedDeviceOperationImpl.writeCharacteristicWithResponse.<anonymous closure> (package:flutter_reactive_ble/src/connected_device_operation.dart:65:39)
E/flutter (12429): #4      _rootRunUnary (dart:async/zone.dart:1362:47)
E/flutter (12429): #5      _CustomZone.runUnary (dart:async/zone.dart:1265:19)
E/flutter (12429): #6      _FutureListener.handleValue (dart:async/future_impl.dart:152:18)
E/flutter (12429): #7      Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:704:45)
E/flutter (12429): #8      Future._propagateToListeners (dart:async/future_impl.dart:733:32)
E/flutter (12429): #9      Future._completeWithValue (dart:async/future_impl.dart:539:5)
E/flutter (12429): #10     _completeOnAsyncReturn (dart:async-patch/async_patch.dart:254:13)
E/flutter (12429): #11     MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart)

remonh87, 2021-03-04:
hmm can you try to use a ble explorer app like nrf connect and check if you can write the value there?  https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp  Problem is that this unfortunately does not provide me enough context to help you further. At least if writing on nrf connect fails as well we know the problem is in the peripheral.

realrk69, 2021-03-05:
I had this problem too. Turns out this is a problem with some devices only. Probably a vendor specific problem which prevents discovery of characteristics (in my case on some infinix, xiaomi and oneplus phones). Never faced this problem on Android 6+ on Samsung, iPhone or other manufacturers, and this only happens with NRF chips. Even the NRF default app does not discover or write to the characteristics, so I guess this library is not the issue.

bobosette, 2021-03-05:
Mmm this could be interesting. For me it happens only on an old Huawei.

closed 2021-04-27 by remonh87