PhilipsHue/flutter_reactive_ble #236: Flutter 2.0 ios compile error SwiftProtobuf

HelloBytes, 2021-03-05:
When compiling the following error occurs:

/ios/Classes/BleData/bledata.pb.swift:11:8: error: compiling for iOS 11.0, but module 'SwiftProtobuf' has a minimum deployment target of iOS 12.0:build/ios/Debug-iphoneos/SwiftProtobuf/SwiftProtobuf.framework/Modules/SwiftProtobuf.swiftmodule/arm64-apple-ios.swiftmodule
    import SwiftProtobuf

Flutter version 2.0.1 at /Applications/flutter
    • Framework revision c5a4b4029c (19 hours ago), 2021-03-04 09:47:48 -0800
    • Engine revision 40441def69
    • Dart version 2.12.0

werediver, 2021-03-05:
Please, check which version of SwiftProtobuf is specified in you "Podfile.lock".

Looks weird. "flutter_reactive_ble" uses deployment target iOS 11.0, SwiftProtobuf seems to require 9.0 (as per [SwiftProtobuf.podspec](https://github.com/apple/swift-protobuf/blob/main/SwiftProtobuf.podspec)).

HelloBytes, 2021-03-05:
Thanks for the fast reply!

- flutter_reactive_ble (0.0.1):
    - Flutter
    - Protobuf (~> 3.5)
    - SwiftProtobuf (~> 1.0)

- SwiftProtobuf (1.12.0)

werediver, 2021-03-05:
As you can see on the repository's front page (the [README.md](https://github.com/PhilipsHue/flutter_reactive_ble/blob/fc94bb468766339edff2e250c771fdefc24e729b/README.md) file), the latest "flutter_reactive_ble" version is [2.7.3](https://github.com/PhilipsHue/flutter_reactive_ble/releases/tag/2.7.3). Please switch to the latest version and try again.

HelloBytes, 2021-03-05:
I use the latest version. the 0.0.1 is I think from https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/ios/flutter_reactive_ble.podspec

HelloBytes, 2021-03-05:
`pod update` in the ios folder fixed the issue, thank you for your time!

closed 2021-03-05 by HelloBytes