PhilipsHue/flutter_reactive_ble #238: Example app does not connect

patolax, 2021-03-07:
**bug**
I am running this code to create a GATT server on my PC.
https://github.com/microsoft/Windows-universal-samples/tree/master/Samples/BluetoothLE/cs
Example app lists my PC after scanning. However when I try to connect to it it stops at "connecting" status. 

[image]

**To Reproduce**
Steps to reproduce the behavior:
```
D/        (15732): Gralloc Register  w:720, h:1464, f:0x1, usage:0xb00, ui64Stamp:343342, sSize:4239360, line = 2106
W/ctivebleexampl(15732): Accessing hidden field Ljava/nio/Buffer;->address:J (light greylist, reflection)
D/BluetoothAdapter(15732): isLeEnabled(): ON
D/BluetoothLeScanner(15732): onScannerRegistered() - status=0 scannerId=11 mScannerId=0
D/        (15732): Gralloc Register  w:720, h:1464, f:0x1, usage:0xb00, ui64Stamp:343388, sSize:4239360, line = 2106
D/BluetoothAdapter(15732): isLeEnabled(): ON
D/BluetoothGatt(15732): connect() - device: 7D:27:FE:E6:D2:3E, auto: true
D/BluetoothGatt(15732): registerApp()
D/BluetoothGatt(15732): registerApp() - UUID=0792c4ea-5a8b-46d5-8173-43e1fc21dd4d
D/BluetoothGatt(15732): onClientRegistered() - status=0 clientIf=11
```

**Expected behavior**
Connect to the GATT server and discover services 

**Smartphone / tablet**
 - Android device 
- https://www.auditech.com.au/telstra-zte-essential-pro-24gx-blue-tick-32gb2gb-dark-green

**Peripheral device**
- Laptop
- GATT server https://github.com/microsoft/Windows-universal-samples/tree/master/Samples/BluetoothLE/cs

remonh87, 2021-03-08:
@patolax  it looks like there is something wrong with your gatt-server. Your logs are quite clear (thanks for adding them): the last line showed `D/BluetoothGatt(15732): onClientRegistered() - status=0 clientIf=11` which is the connecting . What I think  is wrong is that the connectionparameters are rejected by the Android os. Check the connection interval that is specified, android rejects intervals below <11ms.

patolax, 2021-03-09:
I evaluated the GATT server using Lite Blue (https://play.google.com/store/apps/details?id=com.punchthrough.lightblueexplorer) and competing lib (https://github.com/Polidea/FlutterBleLib). Both apps connected to the GATT server and was able to read and wirte characterises. It has to be a implementation issue of  flutter_reactive_ble.

remonh87, 2021-03-09:
can you give me the code snippet that you are using for the connect ?

patolax, 2021-03-09:
Its not my code. I ran your sample without any modification. Only change I did was to the UI to arrange buttons in a Row. See screenshot in above post.

remonh87, 2021-03-10:
The problem is that based on the logs the library is exactly doing what happens within `bluetoothgatt` (android class [link](https://developer.android.com/reference/android/bluetooth/BluetoothGatt) ) . It can be that there is a bug in the underlying library of polidea that we are using but based on the current information I cannot judge this.  What can be the issue is that your peripheral does not conenct with `autoconnect`, this is a know issue on Android. What you can do is specify a timeout in your connectionparameters like: `    _connection = _ble.connectToDevice(id: deviceId, connectionTimeout: Duration(seconds: 20)).listen(....`

patolax, 2021-03-12:
that worked, thanks

closed 2021-03-12 by patolax