PhilipsHue/flutter_reactive_ble #239: iOS Target Deployment

LEMUSADR000, 2021-03-08:
What was the rationale for using a deployment target of 11? My project shouldn't have any reason meeting that, especially with the aggression on iOS to update the OS. Mostly just didn't see any obvious reason for it.

werediver, 2021-03-08:
flutter_reactive_ble sets the deployment target to iOS 11.0, SwiftProtobuf to iOS 9.0 ([apple/swift-protobuf/SwiftProtobuf.podspec](https://github.com/apple/swift-protobuf/blob/e432e3d1da3c123c135935569e01810ac729701b/SwiftProtobuf.podspec)). Why so conservatively? The reason is the same for every library that is consumed by more than one application: a library should not overconstraint its consumers. A library should require whatever it takes to implement its functionality, but no more.

We don't use any features that are not available in iOS 11.0, but your application can set the deployment target much higher, if required.

LEMUSADR000, 2021-03-08:
I'm working on porting over a project and this library is probably the best BLE library I've used so far. In the native project we were able to use minimum deployment target of 9.0 so I was curious what about this project requires 11.0. It's no problem at all for me to 'bump' up to 11, like I said.

closed 2021-03-08 by LEMUSADR000

werediver, 2021-03-09:
@LEMUSADR000 It seems I misunderstood you. I thought you are unhappy with iOS 11.0 deployment target of this library, because it's _too low_, hence my answer explains why we didn't choose a _higher_ deployment target.

I can't answer this from the top of my head. If you're interested, one way to check it is to try to build the example app for a lower iOS version (10.0, 9.0) with a local copy of the library with a modified "Podspec" (use [dependency_overrides](https://dart.dev/tools/pub/dependencies#dependency-overrides) pubspec section).