PhilipsHue/flutter_reactive_ble #240: Reconnect to a disconnected device (a device that is advertising)

LEMUSADR000, 2021-03-09:
I can't quite seem to figure out how to elegantly handle reconnects. What is the 'preferred' way to reconnect to a disconnected device that is advertising after disconnecting?

realrk69, 2021-03-10:
Do you want this to happen automatically? It really depends on your use case which can be more specific. You can create an offline db or use shared preferences to store the previously connected devices to make them appear in the list of paired devices. Otherwise re-scanning and connecting is the only option. If you're tending towards remembering the device ID and mac address for **automatic** reconnection, keep in mind that it might cause issues with multiple devices having similar ID/mac or some security issues too.

LEMUSADR000, 2021-03-10:
@realrk69 Not automatically, no. My use case is that the BLE devices I am interfacing with allow firmware updates. After the completion of a firmware update the device will disconnect, reboot, and then begin advertising. I was just trying to figure out if there was a clean way of reconnecting to a device after this happens other than running through the same connection loop.

LEMUSADR000, 2021-03-12:
It seems that running through the same exact sequence as was done to originally connect a device works just fine consistently, closing. Thank you for your help!

closed 2021-03-12 by LEMUSADR000