PhilipsHue/flutter_reactive_ble #246: Add Linux desktop support

robert-ancell, 2021-03-11:
Flutter desktop support is now in beta, and it would be great if flutter_reactive_ble could work on both mobile and desktop. I've developed a [bluez package](https://pub.dev/packages/bluez) that interfaces with the Linux BlueZ Bluetooth stack, and I'd like to propose we add support for that.

robert-ancell, 2021-03-11:
I opened this issue to split out from https://github.com/PhilipsHue/flutter_reactive_ble/issues/112 as the Windows/MacOS solutions are likely to be quite different.

remonh87, 2021-08-17:
@robert-ancell  it took a while but I migrated the federated plugin structure. Is there any chance you can help out with the linux support for our plugin and modify the pr?

robert-ancell, 2021-08-18:
Thanks @remonh87 - I'll have a go at updating this branch to the new format.