PhilipsHue/flutter_reactive_ble #247: writeCharacteristicWithResponse throws an exception if the response does not contain any data

methompson, 2021-03-11:
**Describe the bug**
I'm currently using a Bluetooth Thermometer in an application. The thermometer has a feature that ultimately locks the thermometer from further temperature reporting until you write a byte of data to a specific characteristic. This characteristic has the property Write With Response, but does not have the write without response property.

When I attempt to write to this characteristic, the writeCharacteristicWithResponse method throws an exception with the following error: "Null check operator used on a null value"

Steps to reproduce the behavior:
The following code produces the error:

`
var stabilizationCharacteristic = QualifiedCharacteristic(
  characteristicId: stabilizationWriteCharacteristicUUID,
  serviceId: stabilizationServiceUUID,
  deviceId: deviceId,
);

var btInst = FlutterReactiveBle();

try {
  await btInst.writeCharacteristicWithResponse(stabilizationCharacteristic, value: l);
} catch (e) {
  print(e);
}
`

In this case, the values passed to QualifiedCharacteristic are defined elsewhere. The code works fine in the older version without null safety.

**Expected behavior**
The expectation is that writeCharacteristicWithResponse would not throw an exception.

**Smartphone / tablet**
 - Device: iPhone 7
 - OS: iOS 14.4
 - Package version: 3.0.0-nullsafety.0

I think I've tracked the problem to ./lib/src/model/result.dart line 15. The variable 'value' is unwrapped without a null check. In this case, because the peripheral doesn't return any value, the variable 'value' can be null.

werediver, 2021-03-12:
First, I want to make sure you understand the meaning of `-Response` in `writeCharacteristicWithResponse` / `writeCharacteristicWithoutResponse`: it's not about a response as you know it from HTTP (where the request is a "question" and the response is an "answer"), it just means "acknowledgment" ("request reception acknowledged").

Then on [Result.dematerialize()](https://github.com/PhilipsHue/flutter_reactive_ble/blob/fc94bb468766339edff2e250c771fdefc24e729b/lib/src/model/result.dart#L15). The thing is, before null-safety `Result.success(·)` was accepting nulls. Not it may be possible, if the class is specialized with an optional type `Result<X?, Y>`, if Dart allows that (have to check).

Good to have some input on the null-safe version, thanks!

werediver, 2021-03-12:
Checked in DartPad with null-safety: Dart does treat optionality as a part of a type and allows specializing generics with optional types just fine, for example:

```dart
void main() async {
  print(Result<String>("xxx").getValue());
  print(Result<String?>("xxx").getValue());
  print(Result<String?>(null).getValue());
}

class Result<Value> {
  const Result(this._value);
  
  Value getValue() => _value;
  
  final Value _value;
}
```

Which means we should consider either specializing `Result` with an optional type where needed, or (which might be better) making sure no nulls are appearing where they are not expected. I suspect in this case a null is appearing where an empty list should.

werediver, 2021-03-12:
@methompson Could check whether the same is happening on Android?

remonh87, 2021-03-12:
I ran into a similar issue and fixed this when extending the example app. Can you check if you encounter the same when you use the `null-safety` branch directly?

methompson, 2021-03-12:
@werediver While I admit I'm not the most experienced with BT LE, you'll note that the code snippet above isn't expecting a value per se. That is, the code is just waiting for the operation to finish. The expectation is that the operation would finish without throwing an exception.

I can't speak any further as to how the source should operate. I've only begun to use the plugin earlier this week.

That said, I tried the plugin on an Android device (Google Pixel 4) and the code snippet above also threw the same exception.

@remonh87 I'm using version `^3.0.0-nullsafety.0` in my pubspec.yaml file. I can't update my application to sdk >=2.12.0 yet because the vast majority of my application isn't set up for null safety yet.

XMitja, 2021-03-18:
I'm getting the same on android using 3.0.0, happens with write both with and without reponse, it doesn't really affect anything other than pollute the logs since I'm not expecting a response anyway
without:
`
[ERROR:flutter/lib/ui/ui_dart_state.cc(186)] Unhandled Exception: Null check operator used on a null value
    #0      Result.dematerialize.<anonymous closure> (package:flutter_reactive_ble/src/model/result.dart:16:34)
    #1      Result.iif (package:flutter_reactive_ble/src/model/result.dart:35:21)
    #2      Result.dematerialize (package:flutter_reactive_ble/src/model/result.dart:15:28)
    #3      ConnectedDeviceOperationImpl.writeCharacteristicWithoutResponse.<anonymous closure> (package:flutter_reactive_ble/src/connected_device_operation.dart:73:39)
    #4      _rootRunUnary (dart:async/zone.dart:1362:47)
    #5      _CustomZone.runUnary (dart:async/zone.dart:1265:19)
    #6      _FutureListener.handleValue (dart:async/future_impl.dart:152:18)
    #7      Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:704:45)
    #8      Future._propagateToListeners (dart:async/future_impl.dart:733:32)
    #9      Future._completeWithValue (dart:async/future_impl.dart:539:5)
    #10     _completeOnAsyncReturn (dart:async-patch/async_patch.dart:254:13)
    #11     MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart)
    <asynchronous suspension>
`
with:
`
[ERROR:flutter/lib/ui/ui_dart_state.cc(186)] Unhandled Exception: Null check operator used on a null value
    #0      Result.dematerialize.<anonymous closure> (package:flutter_reactive_ble/src/model/result.dart:16:34)
    #1      Result.iif (package:flutter_reactive_ble/src/model/result.dart:35:21)
    #2      Result.dematerialize (package:flutter_reactive_ble/src/model/result.dart:15:28)
    #3      ConnectedDeviceOperationImpl.writeCharacteristicWithResponse.<anonymous closure> (package:flutter_reactive_ble/src/connected_device_operation.dart:64:39)
    #4      _rootRunUnary (dart:async/zone.dart:1362:47)
    #5      _CustomZone.runUnary (dart:async/zone.dart:1265:19)
    #6      _FutureListener.handleValue (dart:async/future_impl.dart:152:18)
    #7      Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:704:45)
    #8      Future._propagateToListeners (dart:async/future_impl.dart:733:32)
    #9      Future._completeWithValue (dart:async/future_impl.dart:539:5)
    #10     _completeOnAsyncReturn (dart:async-patch/async_patch.dart:254:13)
    #11     MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart)
    <asynchronous suspension>
`

remonh87, 2021-03-18:
@XMitja  I saw this bug on Android as well. This will be fixed when #248  is merged

remonh87, 2021-03-27:
This should be fixed in version `3.0.0` which is published today

closed 2021-03-27 by remonh87