PhilipsHue/flutter_reactive_ble #250: IOS

HelloBytes, 2021-03-15:
When playing with the bluetooth connection (enabling / disabling) couple of times on IOS this error occurs and the app crashes:

IOS 14
flutter_reactive_ble 2.7.3
Flutter 2.0

Assertion failed: file flutter_reactive_ble/ConnectTaskController.swift, line 16
Assertion failed: file flutter_reactive_ble/ConnectTaskController.swift, line 16
* thread #1, queue = 'com.apple.main-thread', stop reason = Assertion failed
    frame #0: 0x0000000192681e08 libswiftCore.dylib`_swift_runtime_on_report
libswiftCore.dylib`_swift_runtime_on_report:
->  0x192681e08 <+0>: ret    
libswiftCore.dylib`_swift_reportToDebugger:
    0x192681e0c <+0>: b      0x192681e08               ; _swift_runtime_on_report
libswiftCore.dylib`_swift_shouldReportFatalErrorsToDebugger:
    0x192681e10 <+0>: adrp   x8, 347836
    0x192681e14 <+4>: ldrb   w0, [x8, #0x7c8]
Target 0: (Runner) stopped.

werediver, 2021-03-15:
#205 looks distantly related, though the failure is not the same.

werediver, 2021-03-15:
Looking at the error message it seems that your app tries to connect to the same device while it's already being connected. The library may still have a flaw.

What exactly do you mean by "playing ... (enabling / disabling)"? Banging the toggle till the app crashes?

HelloBytes, 2021-03-15:
When I disable or enable the bluetooth, or when I am out of range and there is a timeout, I try to reconnect to the device every xx seconds. It looks like when you try to connect multiple times this error occurs. Before I connect I check if there is a connection and  and like the example disconnect before connect like this:

if (_connection != null) {
      await _connection.cancel();
    }

marcel-bluestone, 2021-03-16:
I have a similar crash experience, trying to subscribe to a notification stream after the connection has been established. (no toggling of bluetooth here!).
 
I am on iOS 14
Flutter 2.1
flutter_reactive_ble 2.7.3

Here's my crash report:

```
Assertion failed: file flutter_reactive_ble/PluginController.swift, line 103
Assertion failed: file flutter_reactive_ble/PluginController.swift, line 103
* thread #1, queue = 'com.apple.main-thread', stop reason = Assertion failed
    frame #0: 0x00000001ac4afff8 libswiftCore.dylib`_swift_runtime_on_report
libswiftCore.dylib`_swift_runtime_on_report:
->  0x1ac4afff8 <+0>: ret    
libswiftCore.dylib`_swift_reportToDebugger:
    0x1ac4afffc <+0>: b      0x1ac4afff8               ; _swift_runtime_on_report
libswiftCore.dylib`_swift_shouldReportFatalErrorsToDebugger:
    0x1ac4b0000 <+0>: adrp   x8, 344158
    0x1ac4b0004 <+4>: ldrb   w0, [x8, #0x998]
Target 0: (Runner) stopped.
```

HelloBytes, 2021-03-16:
> I have a similar crash experience, trying to subscribe to a notification stream after the connection has been established. (no toggling of bluetooth here!).
>
> I am on iOS 14
> Flutter 2.1 [...]

Toggling does not cause this issue sorry for the misconception, I have changed the title. In my code I try to reconnect after it disconnects or fails or when bluetooth is ready (again).

marcel-bluestone, 2021-03-16:
@HelloBytes thanks for the clarification. This seems to be a bug to me. My dart code does not seem to be the cause of the crash, as I don't have any problems with it on Android...

werediver, 2021-03-16:
> Toggling does not cause this issue sorry for the misconception, I have changed the title. In my code I try to reconnect after it disconnects or fails or when bluetooth is ready (again).

@HelloBytes Than you for clarification, it does make a difference.

@marcel-bluestone Even though your issue may be "structurally" similar, your stack-trace is different and the failing operation is different. I'd suggest you to open a separate ticket for it (or reference an existing one).

werediver, 2021-03-16:
@HelloBytes Forgot to mention. _As a workaround_, you can try introducing a delay before you are trying to reconnect (300–500 ms). Please, let us know if that makes any difference.

HelloBytes, 2021-03-16:
> @HelloBytes Forgot to mention. _As a workaround_, you can try introducing a delay before you are trying to reconnect (300–500 ms). Please, let us know if that makes any difference.

I can try it tomorrow when I work on the project!

HelloBytes, 2021-03-23:
It's look like it will only occur when you try to connect multiple times in a short time window. So now when connecting I set a flag that it is connecting. When a second attempt tries to connect and the flag is still connecting it will skip that attempt.

calebisstupid, 2021-08-12:
I'm getting the same thing (along with https://github.com/PhilipsHue/flutter_reactive_ble/issues/269) when doing the following:

1. enable ble pairing on device
2. scan for devices in my code
3. find and connect to device
4. re-open app and re-enable ble pairing on device
5. scan for device in my code
6. find and (attempt to) connect to device (this fails due to timeout)
7. after timeout, attempt to connect to device again and I get this crash

remonh87, 2021-08-13:
@calebisstupid  I think it is best you add a small timeout between requests. Like mentioned above. BLE does not tend to appreciate subsequent request in a short period of time.

calebisstupid, 2021-08-13:
> @calebisstupid I think it is best you add a small timeout between requests. Like mentioned above. BLE does not tend to appreciate subsequent request in a short period of time.

In my testing, it doesn't matter if I wait seconds or an entire day. As long as my device is showing as a previous device in the iOS bluetooth settings, it will always timeout when attempting to reconnect.

bretmh, 2021-09-21:
Before reconnecting close your existing stream to your previous connection.

AnimeshHestabit, 2022-04-20:
IOS app is crashed after the connecting with device so for handle the crash using **scanStream.cancel()**, but after doing this one of my readCharacteristic is not working fine i am not getting the battery status from device, but this is working fine with Android. According to me this error is because of **scanStream.cancel()** because in IOS app we used this and Android we did not use this.

I am sharing the readCharacteristic method which is not working in IOS, please help me out into this.

 version : flutter_reactive_ble: ^5.0.2

```
 if (Platform.isIOS) {
       connectedBluetooth.value = true;
            connectionForData();
            _scanStream.cancel();

   }
```
  Future<void> connectionForData() async {
    try {
 flutterReactiveBle
              .readCharacteristic(characteristicBattery)
              .then((List<int> result) {
            debugPrint("HERE ==>> connected bluetooth then");
            for (var i in result) {
              ///read the battery status (byte code)
              debugPrint("HERE ==>> connected bluetooth for");
              batteryStatus(i);
            }
          });
}, onError: (Object error) {
   
        CommonUtils.toastMesg(error.toString());
        // Handle a possible error
      });
    } catch (e) {
      debugPrint("HERE ==>> ${e.toString()}");
    }