PhilipsHue/flutter_reactive_ble #252: Error build for iOs with codemagic

ffamar, 2021-03-27:
I'm trying to build my app using codemagic but I receive the following error:

_Failed to build iOS app

Error output from Xcode build:

↳

** BUILD FAILED **

Xcode's output:

↳

/Users/builder/clone/ios/Runner/GeneratedPluginRegistrant.m:10:9: fatal error: module 'flutter_reactive_ble' not found

@import flutter_reactive_ble;

~~~~~~~^~~~~~~~~~~~~~~~~~~~

1 error generated.

note: Using new build system

note: Building targets in parallel

note: Planning build

note: Constructing build description
_

Today I switched to version 3.0 null safety of flutter_reative_ble, everything run weel in android.
Can anyone help me?
Thanks!

remonh87, 2021-03-28:
This is a known problem in Flutter (had this myself as well in the past). I would checkout your iOS config. These flutter issues provide you some tips:  [issue1 ](https://github.com/flutter/flutter/issues/43986#)  [Issue2](https://github.com/flutter/flutter/issues/53573). Hope this helps

ffamar, 2021-03-28:
Thanks for the help @remonh87 !

My problem was a  different platform version beetween Podfile and ios/Runne.xcodeproj/project.pbxproj

I solved by changing the value
S_DEPLOYMENT_TARGET=11.0
 in project.pbxproj as declared in Podfile.

closed 2021-03-28 by ffamar