PhilipsHue/flutter_reactive_ble #253: Support for printing with thermal printers.

wisnuwiry, 2021-04-02:
**Describe the bug**
When I want to send data to a bluetooth printer device (Thermal Printer) with the sending method `writeCharacteristicWithoutResponse` or` writeCharacteristicWithResponse`, an error will occur and the data will not be sent.

Exception:

```java
Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "GATT exception from MAC address 66:22:E9:C7:8D:74, with type BleGattOperation{description='CHARACTERISTIC_WRITE'}")
```

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one device thermal printer
2. Write data with `writeCharacteristicWithoutResponse` or `writeCharacteristicWithResponse`
3. An error will occur

**Expected behavior**

Can run perfectly without any bugs, when printing with a thermal printer.

**Smartphone / tablet**
 - Device: Android Redmi
 - OS:  Android 8
 - Package version: 3.0.0

**Peripheral device**
- Vendor, model: GOOJPRT PT.2.1.0 (Portable Printer)
- Does it run a custom firmware / software: no

**Additional context**

remonh87, 2021-04-02:
@wisnuwir thank you for the clear description. Normally a gatt exception is an error between the OS and the peripheral. Can you try to write a value using a Bluetooth explorer like nrf connect [link](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en&gl=US) ? I think you will run into the same issue there as well.

wisnuwiry, 2021-04-02:
Yes I have the same problem. Is there a solution to my problem?

I have used several libraries in pub.dev for print to printer devices. But the results couldn't be satisfactory. I would like to combine reactive ble with send data in the printer, which might be more satisfying.

remonh87, 2021-04-02:
I think there is nothing with the library but something is off with either your peripheral (in that case I would checkout docs of the vendor) or with the phone. There are two things you can check:

1. Make sure you are sending correct data to the peripheral
2. Try to write data to the printer using another phone

remonh87, 2021-04-07:
closed for now since this is not related to our library but more peripheral or phone issue.

closed 2021-04-07 by remonh87