PhilipsHue/flutter_reactive_ble #258: Use advertised service UUIDs (iOS)

flo80, 2021-04-13:
When scanning for devices, the `serviceData` of the device is always (see note 1) empty. 
This seems to be intended (see also issue #198) but means, to know which service are advertised, one has to explicitly scan for services on the device. 

Note 1: it seems like some service data  gets picked up, e.g. exposure notification 

    ID: EC58332D-90DA-7A3C-82F3-7F420425815B 
    Services: {fd6f: [24, 27, 255, 218, 56, 245, 11, 114, 173, 241, 138, 84, 51, 196, 27, 212, 181, 93, 196, 124]}

This can easily be tested eg. in the example app by printing service data of scan results; in ble_scanner.dart, line 30 add:

    if(device.serviceData.isNotEmpty)
      print('found device ${device.name} ${device.id} with services ${device.serviceData}');

*Alternative*
At least on iOS, it is possible to get the advertised service UUIDs which would allow to eliminate the need for a scan (unsure about the situation on Android). 

*Background*
In the context I am using flutter_reactive_ble, we do not know the exact UUID of the advertised service (i.e. cannot scan for it) but we can filter the scan results by checking parts of the UUID. 

*Solution*
Instead of using `CBAdvertisementDataServiceDataKey` use `CBAdvertisementDataServiceUUIDsKey` (see attached 
[patch.txt](https://github.com/PhilipsHue/flutter_reactive_ble/files/6303337/patch.txt)

*Result*
Instead of just getting no service data from scanning, the initial scan already yields advertised service UUIDs, e.g.

    ID: BE8CF775-DF59-D5DB-3457-6AADD9C77017
    Services: {1bb51700-58b9-1411-82fb-8dd19e64568a: [27, 181, 23, 0, 88, 185, 20, 17, 130, 251, 141, 209, 158, 100, 86, 138]}

remonh87, 2021-04-13:
This is a good addition to the scan results. I also see this is possible on Android. https://developer.android.com/reference/android/bluetooth/le/ScanRecord

xvrh, 2021-04-15:
I would like this feature too.
@flo80 do you intend to submit a pull request? If not I may give it a try.

xvrh, 2021-04-15:
I went ahead and implemented the feature for iOS and Android in #260. Let me know if there is any concern.

flo80, 2021-04-20:
@xvrh sorry for not replying earlier, I have been too busy the last week. I have seen #260 and it looks good to me, thanks for implementing it!

PR #260 (2021-05-02): Add serviceUuids in ScanResult
Fixes #258 

Add `List<Uuid> serviceUuids` in `DiscoveredDevice`
closed 2021-04-20 by flo80