PhilipsHue/flutter_reactive_ble #259: Receive read characteristic notifies while iOS is minimized.

safield, 2021-04-14:
My app receives read characteristic notifies from a bluetooth device at regular 2 second intervals.

When the app is minimized on Android, those notifies continue to propagate to my app. When on iOS these notifies immediately cease once the app is minimized.

Is there a way to receive notifies in the background on iOS, or is this simply a limitation of the iOS platform?

remonh87, 2021-04-24:
@werediver  can you answer this question? think it should work normally.

werediver, 2021-04-28:
@safield Please refer to [Core Bluetooth Background Execution Modes](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW7). You may need to declare `bluetooth-central` background execution mode, if you want to process notifications when your app is minimized.

I assume this resolves the question.

closed 2021-04-28 by werediver

roberoad, 2021-11-08:
@safield did this work for you? I need to get continued updates from my BLE device when app is not in foreground and I'm struggling to find anyone successfully doing this with Flutter.

jawadabbasi923, 2023-02-28:
@safield have you solved it ?