PhilipsHue/flutter_reactive_ble #261: Unable to send messages over 20 bytes

sausage-dog, 2021-04-21:
Unable to send messages over 20 bytes with  _await _ble.writeCharacteristicWithResponse()_.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one BLE device
2. Negotiated higher MTU size
2. Subscribe for notifications
2. Write to a characteristic 

**Expected behavior**
I expect to be able to send messages over 20 bytes in a single message.

Messages at 20 bytes or below work fine.

Messages above 20 bytes crash and give me

> _Exception (Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "GATT exception from MAC='XX:XX:XX:XX:XX:XX', status 133 (GATT_ERROR), type BleGattOperation{description='CHARACTERISTIC_WRITE'}. (Look up status 0x85 here https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h)"))

Device listening on is capabale of receiving more than 20 bytes via BLE scanner or flutter blue (though for flutter blue I am not able to **receive** more than 20 bytes but able to send over 20 bytes...).

**Smartphone / tablet**
 - Device: Galaxy Note 10
 - OS: Android 11
 - flutter reactive BLE version: 3.0.0

**Peripheral device**
- Raspberry pi 4
- Running Bleno

remonh87, 2021-04-24:
The error that you are showing is that there is mostly that the write is rejected by the peripheral itself. You should be able to send more then 20bytes  by negotiating a higher MTU (use `negotiaMTU` operation for this) with your Raspberry pi device.  However it can happen that your peripheral does not support a higher MTU and then you need to chop your message in chunks of 20 bytes.

closed 2021-04-28 by remonh87

sausage-dog, 2021-04-28:
I did chop the message, but the device definitely can receive longer messages as I can do so with other libraries.

remonh87, 2021-04-28:
I really would try negotiating a higher MTU and see if this works. I have used this library for sending data over 20bytes for multiple projects without problems. Also see if the issue can be reproduced on another phone if possible. The error is yielded by the Android operating system