PhilipsHue/flutter_reactive_ble #264: Can't find (any) devices when scanning with services ids

benoit-rolandeau-act, 2021-04-29:
Hi everybody!

**Describe the bug**
Can't find devices when giving services ids.

**To Reproduce**
When I use the method `scanForDevices` with `withServices` parameter equals to `[]`:

```
_scanSub = _flutterBle.scanForDevices(
        withServices: [],
      ).listen(_addScanDeviceCallback);
```

I detect three devices around me (included mine which its serial number is `AA:AA:AA:AA:AA:AA` (I changed it for the logs)):

```
I/flutter (32034): 2021-04-29T10:34:00.096255-[debug]: Device discovered: DiscoveredDevice(id: 25:BD:17:F2:38:87, name: , serviceData: {}, manufacturerData: [6, 0, 1, 9, 32, 2, 39, 222, 119, 30, 141, 69, 240, 20, 17, 5, 163, 148, 15, 192, 54, 37, 162, 35, 40, 174, 79, 73, 4], rssi: -23)
I/flutter (32034): 2021-04-29T10:34:00.102196-[debug]: Device discovered: DiscoveredDevice(id: 0B:2F:8E:09:59:77, name: , serviceData: {}, manufacturerData: [6, 0, 1, 9, 32, 2, 204, 48, 151, 165, 7, 254, 12, 243, 127, 24, 227, 181, 226, 35, 175, 60, 13, 148, 197, 138, 99, 23, 132], rssi: -28)
I/flutter (32034): 2021-04-29T10:34:05.811787-[debug]: Device discovered: DiscoveredDevice(id: AA:AA:AA:AA:AA:AA, name: MyDevice, serviceData: {}, manufacturerData: [], rssi: -36)
```

But when I give a serviceId (which is managed by the device I want to discover): 

```
_scanSub = _flutterBle.scanForDevices(
        withServices: [Uuid.parse("00001800-0000-1000-8000-00805f9b34fb")],
      ).listen(_addScanDeviceCallback);
```

No devices are found.

I know that my device has this service (and the lib can detect it), because when I use the "without restrictions" scan and ask for discovering services:
`_flutterBle.discoverServices`

The service: "00001800-0000-1000-8000-00805f9b34fb" is on the list returned

**Expected behavior**
When I set one of the services in the discover method I expect to find my device.

I have also tested, giving all the services managed by the device to the scanMethod parameter. And also try to test with other specific service id.

**Smartphone / tablet**
 - Device: Samsung Galaxy S10+
 - OS: Android 11
 - Package version: 3.0.0

**Peripheral device**
- Vendor, model: CUSTOM
- Does it run a custom firmware: yes

**Additional context**
None

Thanks for your help!

remonh87, 2021-04-29:
I think you need to double check with the vendor if the service is discoverable which means it is part of the ble advertisement data. There is a difference between a services being part of the GATT or  if it is being advertised.  For more explanation on this topic I found this article quite useful: https://www.novelbits.io/bluetooth-low-energy-advertisements-part-1/ .

benoit-rolandeau-act, 2021-04-29:
Oh, great I understand. 

I think you're right: I downloaded `LightBlue` and seen in the "advertised services UUIDs", one service (which I don't know) but not those I wanted. 

Therefore, I understand why I don't find the devices.

Thanks for your help and sorry for the inconvenience!

Have a good day

closed 2021-04-29 by benoit-rolandeau-act

remonh87, 2021-04-29:
No problem btw we will soon provide a new version ourselves that discloses the serviceuuids