PhilipsHue/flutter_reactive_ble #268: how to check if device is connected before subscribing

vontdeux, 2021-05-06:
**Describe the bug**
A clear and concise description of what the bug is.
I'm trying to modify what the example included in the source code, so that onTap of list of devices, would connect and subscribeToCharacteristic.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the example on any device
2. on tap of connect button, and right after Device is connected, subscribeToCharacteristic
3. using await wont help on this method call

**Expected behavior**
A clear and concise description of what you expected to happen.
onTap of list of devices, would connect and subscribeToCharacteristic.

**Smartphone / tablet**
 - Device: [e.g. iPhone X] any
 - OS: [e.g. iOS 13, Android 10] any
 - Package version: [e.g. 1.0.0] any

**Peripheral device**
- Vendor, model: [e.g. Fitbit Versa, iPhone SE, or CUSTOM]
- Does it run a custom firmware - no
**Additional context**
Add any other context about the problem here.

not sure if this is where I ask this question, but i also did ask in stackoverflow
https://stackoverflow.com/questions/67413789/flutter-reactive-ble-how-to-check-if-device-is-connected-before-subscribing

remonh87, 2021-05-06:
@vontdeux  I answered the question on stackoverflow. Hope this helps will close the issue here.

closed 2021-05-06 by remonh87

vontdeux, 2021-05-17:
hi remonh87, i dont want to create a new issue, im not sure how to reach out to you guys.
but i also stumble upon the same issue as this guy (after following your advice for this issue) here
 
 https://stackoverflow.com/questions/67433382/how-to-properly-use-flutter-reactive-ble-writetocharacteristic-and-subscribetoch/67465920#67465920
 
 how to properly use the subscribe and write?

vontdeux, 2021-05-17:
if i use 

```
subcribe.listen({
write(characteristic,data);
})
```

the write will never get executed unless i have another write after the subscribe
and the readfromcharacteristic can only get the value if the subscribe is listening. rendering it useless for my usage.

I need  to be able to write and get the respond from bluetooth module. and the respond is following a custom protocol that I need to dissect the incoming bytes first to be able to determine what is the actual data.

not sure how to properly explain it here.