PhilipsHue/flutter_reactive_ble #269: Unable to reconnect to BLE device after being disconnected

sausage-dog, 2021-05-10:
**Describe the bug**
I managed to connect to a BLE device, write to it and read responses using a callback. 

Now I begun testing what happens when channel gets disconnected at either end.

Code below works fine on the first run, but if I disconnect the device I can not connect to it again.
If I hard reset flutter dev environment it will work again but never in the same session.

Not sure if there is some variable I need to clear or perform other action to reconnect, I could not find an example for reconnecting in the examples provided.

**To Reproduce**

Run the following code to connect to a BLE device, disconnect BLE device, run it again.

```
Future <bool> connectBLE() async{

   log("Connecting to BLE device:");

  _ble.connectToAdvertisingDevice(id:"test",  prescanDuration: Duration(seconds: 5), withServices: [Service_UUID, Characteristic_UUID], connectionTimeout: const Duration(seconds:  10));
 
  // set up length of data being sent
    var mtu = await _ble.requestMtu(deviceId: pi_id, mtu: 250);
  _rxCharacteristic = QualifiedCharacteristic(serviceId: Service_UUID, characteristicId: Characteristic_UUID, deviceId: pi_id);
  Stream<List<int>> _receivedDataStream = _ble.subscribeToCharacteristic(_rxCharacteristic);  
  
  log("Negotiated size: ${mtu}");  
  
  // handle callbacks from BLE
  _receivedDataStream.listen((rec_data) {          

    // passes raw bytes to reassembler
    processMessage(rec_data);

    // code to handle incoming data
  }, onError: (dynamic error) {
    // code to handle errors
    // reset BLE TODO
  });

  return true;
}
```

On the first run, I get the following messages:
```
[log] Connecting to BLE device:
I/BluetoothAdapter(15466): STATE_ON
D/BluetoothGatt(15466): connect() - device: DC:A6:32:E1:7C:0A, auto: true
I/BluetoothAdapter(15466): isSecureModeEnabled
D/BluetoothGatt(15466): registerApp()
D/BluetoothGatt(15466): registerApp() - UUID=ccef16d0-acbc-4969-b7c3-5facd1a3bfd9
D/BluetoothGatt(15466): onClientRegistered() - status=0 clientIf=15
D/BluetoothGatt(15466): onClientConnectionState() - status=0 clientIf=15 device=DC:A6:32:E1:7C:0A
D/BluetoothGatt(15466): configureMTU() - device: DC:A6:32:E1:7C:0A mtu: 250
D/BluetoothGatt(15466): onConnectionUpdated() - Device=DC:A6:32:E1:7C:0A interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(15466): onConfigureMTU() - Device=DC:A6:32:E1:7C:0A mtu=250 status=0
D/BluetoothGatt(15466): configureMTU() - device: DC:A6:32:E1:7C:0A mtu: 250
D/BluetoothGatt(15466): onConfigureMTU() - Device=DC:A6:32:E1:7C:0A mtu=250 status=0
[log] Negotiated size: 250
```

On every subsequent reconnection attempt, none of the above messages appear:

```
[log]Connecting to BLE device:
[log] Negotiated size: 0
I/BluetoothAdapter(20695): STATE_ON
D/BluetoothLeScanner(20695): Start Scan with callback
D/BluetoothLeScanner(20695): onScannerRegistered() - status=0 scannerId=15 mScannerId=0
```

**Expected behavior**
I expect to be able to connect again with the same function and have similar output of underlying BLE operations.

I am able to connect to the BLE periphery using BLE scanner and using other BLE compatible central devices.

**Smartphone / tablet**
 - Samsung Note 10
 - OS: Android 11
 - Package version: flutter_reactive_ble: ^3.0.0

**Peripheral device**
- raspberry Pi + custom hardware
- Running Bleno

remonh87, 2021-05-10:
@sausage-dog thank you for the elaborate description of the bug this helps. One question did you try the same on our example app? I used `connectToAdvertisingDevice `  did subscribe disconnect and reconnect without problems  however I do not have the same phone + peripheral. 

I noticed a couple of interesting things in the logs:
1. You connect using `autoconnect` according to logoutput: `D/BluetoothGatt(15466): connect() - device: DC:A6:32:E1:7C:0A, auto: true`. However in your codesnippet I noticed you specified a connection time-out which should set autoconnect to `false` like in my log output `D/BluetoothGatt(21116): connect() - device: XX:XX:XX:XX:XX:XX, auto: false
` . Are you sure that the log corresponds with the code?

2. I also noticed that on the second output the mtu size is negotiated before the scanner starts. So it looks like that you do not ensure connection to be established before executing the mtu negotiation. I would do sth like: 
```dart
 _connection =  _ble.connectToAdvertisingDevice(id:"test",  prescanDuration: Duration(seconds: 5), withServices: [Service_UUID, Characteristic_UUID], connectionTimeout: const Duration(seconds:  10)).listen(
      (update) {
        if(update.connectionState==DeviceConnectionState.connected){
          //do MTU stuff
        });
```

3. I didn't see a disconnect call in the form of `_connectionSubscription.cancel()`.  I would recommend disconnecting by cancelling the connection stream because this also ensures the streams to be terminated on the native level. See [example app
](https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/example/lib/src/ble/ble_device_connector.dart)
Hopefully this helps

sausage-dog, 2021-05-18:
Thanks for answer.

I rewrote everything in using the format from the unofficial example UART app linked at the bottom.  

It had many things I didn't know where possible or potentially needed from the pub.dev.

Not sure what the issue was but it works now.

remonh87, 2021-05-18:
Cool if you miss some docs feel free to raise a new issue and we will fix this.

closed 2021-05-18 by remonh87

calebisstupid, 2021-08-12:
@sausage-dog any ideas what the fix is? I just swapped to this library from `flutter_blue` hoping it would help the reconnecting issues. In my experience, I am unable to reconnect to my ble device until I manually "forget" the device from the iOS settings.

My code follows the same structure as the above example. And the following is how I reproduce:

1. enable pairing on BLE device
2. scan for devices in my app
3. see and connect to my BLE device
4. close / re-launch my app
5. enable pairing on BLE device
6. scan for devices
7. attempt to connect and it times out. on second attempt to connect I get a full crash (as mentioned in https://github.com/PhilipsHue/flutter_reactive_ble/issues/250)