PhilipsHue/flutter_reactive_ble #278: Use this code from iOS

ozshabat, 2021-05-20:
Hello!

Thanks for your great contribution of this library!

I was wondering if I can use this API to control external hardware of Philips from an iOS app? I saw that you made an iOS code and just wanted to know if I can use it in practice (by following your instructions).
 
Thanks,
Oz.

werediver, 2021-05-20:
This is an open-source project with a permissive license (the three-clause BSD license), so as long as you don't violate it you can use the project in whole or in part for your needs.

Note that this project does not implement any of the Philips Hue / Signify proprietary BLE interfaces and just exposes iOS and Android BLE GATT APIs (in a unified and, perhaps, slightly incomplete form) to Flutter environment.

werediver, 2021-05-20:
I don't know anything about [PhilipsHueSDK-iOS-OSX](https://github.com/PhilipsHue/PhilipsHueSDK-iOS-OSX). Please, refer to the information in that repository. Pay note that it hasn't been updated for five years.

ozshabat, 2021-05-20:
Thank you for your help @werediver

closed 2021-05-20 by ozshabat