PhilipsHue/flutter_reactive_ble #281: Error during build after 3.1.1 update

ffamar, 2021-05-24:
Hi, today I tried to update new version 3.1.1 but I get the following error when I build:

FAILURE: Build failed with an exception.                                
                                                                        
* What went wrong:                                                      
A problem occurred configuring project ':flutter_reactive_ble'.         
> com.android.build.gradle.internal.api.TestVariantImpl_Decorated cannot be cast to com.android.build.gradle.internal.core.InternalBaseVariant
                                                                        
* Try:                                                                  
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
                                                                        
* Get more help at https://help.gradle.org    

Thanks for your help

remonh87, 2021-05-24:
we updated the google dependencies I think you need to update the  gradle-wrapper to `6.8.1`.

ffamar, 2021-05-24:
Yes, the problem was the gradle version.

In file **android/build.gradle** I updated  the gradle version from:

_classpath 'com.android.tools.build:gradle:3.5.0'_

to:

_classpath 'com.android.tools.build:gradle:4.2.1'_

This solved the problem.
Thanks!

closed 2021-05-24 by ffamar