PhilipsHue/flutter_reactive_ble #283: scanForDevices() twice causes crash on iOS

abegehr, 2021-05-28:
**Describe the bug**
I'm calling `scanForDevices()` twice and this causes a crash. See attached screenshot.

**To Reproduce**
Steps to reproduce the behavior:
1. Call `scanForDevices()` twice.

**Expected behavior**
It does not matter how often `scanForDevices()` is called and it does not cause a crash.

**Smartphone / tablet**
 - Device: iPhone 12 Pro
 - OS: iOS 14.6
 - Package version: 3.1.1

**Peripheral device**
not applicable

[image]

Taym95, 2021-05-29:
This assert is being triggered 
https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/ios/Classes/Plugin/PluginController.swift#L153

You need to call `stopScanning` before calling the second `scanForDevices`

abegehr, 2021-06-03:
Thanks for the info!

abegehr, 2021-06-03:
Actually, @Taym95, I'm not finding stopScanning() on the FlutterReactiveBle service: https://pub.dev/documentation/flutter_reactive_ble/latest/flutter_reactive_ble/FlutterReactiveBle-class.html.
I'm on the lastest version 3.1.1+1.

werediver, 2021-06-03:
@abegehr To terminate a scan you need close the stream returned when the scan is started. Same as with a connection: to terminate it you just close the connection status stream returned when the connection was initiated.

werediver, 2021-06-13:
I suppose, the original issue is resolved through explanation. The library API design reflects the underlying BLE API.

closed 2021-06-13 by werediver