PhilipsHue/flutter_reactive_ble #284: GenericFailure<WriteCharacteristicFailure> on Write

ChrisKretschmer, 2021-05-29:
**Describe the bug**

I try to write to a characteristic using writeCharacteristicWithoutResponse or writeCharacteristicWithResponse. Both calls throw an error as follows:

```
E/flutter ( 8164): [ERROR:flutter/lib/ui/ui_dart_state.cc(199)] Unhandled Exception: Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "GATT exception from MAC address 1F:31:XX:XX:XX:XX, with type BleGattOperation{description='CHARACTERISTIC_WRITE'}")
E/flutter ( 8164): #0      Result.dematerialize.<anonymous closure> (package:flutter_reactive_ble/src/model/result.dart:22:13)
E/flutter ( 8164): #1      Result.iif (package:flutter_reactive_ble/src/model/result.dart:34:21)
E/flutter ( 8164): #2      Result.dematerialize (package:flutter_reactive_ble/src/model/result.dart:15:28)
E/flutter ( 8164): #3      ConnectedDeviceOperationImpl.writeCharacteristicWithoutResponse.<anonymous closure> (package:flutter_reactive_ble/src/connected_device_operation.dart:73:39)
E/flutter ( 8164): #4      _rootRunUnary (dart:async/zone.dart:1362:47)
E/flutter ( 8164): #5      _CustomZone.runUnary (dart:async/zone.dart:1265:19)
E/flutter ( 8164): <asynchronous suspension>
E/flutter ( 8164): #6      Bluetooth.sendCommand (package:evo_client/provider/bluetooth.dart:142:5)
E/flutter ( 8164): <asynchronous suspension>
```

The corresponding code looks as follows:

```
  Future<void> sendCommand(BluetoothCommand command) async {
    var data = command.asBytes();
    final characteristic = GetCharacteristic() as QualifiedCharacteristic;
    await flutterReactiveBle.writeCharacteristicWithoutResponse(characteristic, value: [108,0,0,0,0,0,0,0,0,0,0,0]);
  }
```

(I replaced the data with a static array for testing purposes).

The characteristic is 12 Bytes long, so that should not be an issue.

Writing to that Characteristic using BLE Scanner works from that device.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one Arduino Nano BLE Sense with my Sketch from https://github.com/adastra-rocketry/adastra-evo/tree/main/Firmware
2. Write to the Command Characteristic
3. Observe a failure with exception (including the part of the stack trace, belonging to this package) …

**Expected behavior**
The write should work obviously 

- [X] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: Oneplus North N10 5G
 - Android 10  (Oxygen OS 10.5.13.BE89BA)
 - Package version: ?

**Peripheral device**
- Vendor, model: Arduino Nano BLE Sense
- Does it run a custom firmware / software: yes
https://github.com/adastra-rocketry/adastra-evo/tree/main/Firmware

**Additional context**
Add any other context about the problem here.

ChrisKretschmer, 2021-05-29:
I just checked my code with another Device - an Nokia 7.2 - and the problem exists there too.

remonh87, 2021-05-30:
A GATT exception mainly happens in case there is something wrongly configured at the peripheral side. Can you check if you can write using this a bluetooth explorer like:  https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en ?

ChrisKretschmer, 2021-05-30:
Reading and writing using the nRF connect app works.
It also works using a custom web client using webBluetooth in the browser.

It really seams to be something related to Flutter and the Arduinos BLE implementation.

Is there a way to enable a more verbose logging?

ChrisKretschmer, 2021-05-30:
Oh boy... never mind.
I simply wrote to the wrong characteristic :D

Thank you for your support - it all works now. :)

closed 2021-05-30 by ChrisKretschmer

Sabahat06, 2023-05-19:
What's issue with characteristics @ChrisKretschmer