PhilipsHue/flutter_reactive_ble #285: Build fails on 3.1.1 and above

sander121, 2021-05-31:
When building for android with 3.1.0 everything works fine.
When building with 3.1.1and above I get the error:

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':flutter_reactive_ble'.
> class com.android.build.gradle.internal.api.TestVariantImpl_Decorated cannot be cast to class com.android.build.gradle.internal.core.InternalBaseVariant (com.android.build.gradle.internal.api.TestVariantImpl_Decorated is in unnamed module of loader org.gradle.internal.classloader.VisitableURLClassLoader @5db3
98dd; com.android.build.gradle.internal.core.InternalBaseVariant is in unnamed module of loader org.gradle.internal.classloader.VisitableURLClassLoader @69421090)

I do not understand enough of gradle to see the problem. Does someone else have this as well and maybe also found a solution? Thank you.

remonh87, 2021-05-31:
We needed to update to the latest stable gradle version + wrapper. 

If you update tools to  `classpath 'com.android.tools.build:gradle:4.2.1'`  and the wrapper version to 6.8.1. The app should build

sander121, 2021-05-31:
> We needed to update to the latest stable gradle version + wrapper + android dependencies.
> 
> If you update tools to `classpath 'com.android.tools.build:gradle:4.2.1'` and the wrapper version to 6.8.1. The app should build

Thank you for your quick help. This indeed makes the error messages I got before disappear. Unfortunately it causes many more problems, all releated to incompatible kotlin and gradle versions. I will stick to 3.0.0 for now.

closed 2021-05-31 by remonh87

farr64, 2021-10-21:
Please see the note I just made at #410 regarding a comment made by @YellowShark about setting the Android SDK to 31.