PhilipsHue/flutter_reactive_ble #286: iOS crashes when turning on Bluetooth switch at app settings(in system Settings)

flutterBLE, 2021-06-09:
**Describe the bug**
When user leave app to Settings for granting Bluetooth permission to app, app crashes after turning on Bluetooth switch.

**To Reproduce**
Steps to reproduce the behavior:
1. turn off Bluetooth switch in app settings.
2. Run app.
3. Listen ble statusStream.
4. Go to app settings at system Settings, turn on Bluetooth switch.
5. App crashes

Error:
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGKILL
    frame #0: 0x00000001bcb6d908 libsystem_kernel.dylib`mach_msg_trap + 8
libsystem_kernel.dylib`mach_msg_trap:
->  0x1bcb6d908 <+8>: ret
libsystem_kernel.dylib`mach_msg_overwrite_trap:
    0x1bcb6d90c <+0>: mov    x16, #-0x20
    0x1bcb6d910 <+4>: svc    #0x80
    0x1bcb6d914 <+8>: ret
Target 0: (Runner) stopped.
Lost connection to device.

werediver, 2021-06-13:
Quick googling points to a similar report: https://github.com/Baseflow/flutter-permission-handler/issues/293

Which mentions this: https://github.com/Baseflow/flutter-permission-handler/issues/368#issuecomment-763854735

Any chance this is your problem? (check the last link)

flutterBLE, 2021-06-15:
Not open app settings programmatically, I press home button and then click system settings icon, find app settings in list, finally turn on/off Bluetooth switch in app settings.
App crashes in my iphone se 2 / iphone 6s plus. If app had bluetooth permission already, turn off Bluetooth switch in app settings still make app crash.

ky1vstar, 2021-11-02:
It's absolutely normal behaviour for iOS to kill application when you change its permissions in Settings app. This is the same for native apps.

Taym95, 2021-11-15:
Please check this response [`Your app is not crashing...`](https://stackoverflow.com/questions/15930708/having-app-restart-itself-when-it-detects-change-to-privacy-settings/15930922#15930922)
 
 I will close this issue for now because it seems like normal IOS behavior.

closed 2021-11-15 by Taym95

octocat-1, 2021-12-28:
> It's absolutely normal behaviour for iOS to kill application when you change its permissions in Settings app. This is the same for native apps.

Huh? That's rude, what if the user had unsaved data?

octocat-1, 2021-12-30:
This feels like a bug, or an oversight on Apple's side. If it's by design, then it goes against Apple's own advice.
https://developer.apple.com/library/archive/qa/qa1561/_index.html

If setting changes cause an app to close, there should be an alert that communicates that to the user in advance.

remonh87, 2021-12-30:
@octocat-1  I agree it is bad user experience but I think there is little we can do about it. The stack trace is not thrown by our lib but by the OS

GordonHuangYong, 2023-05-30:
> It's absolutely normal behaviour for iOS to kill application when you change its permissions in Settings app. This is the same for native apps.

not really

tabekg, 2024-04-23:
any solutions?