PhilipsHue/flutter_reactive_ble #287: [Android] All connect fails after writing exception

benoit-rolandeau-act, 2021-06-09:
**Describe the bug**
I have only tested on Android smartphone. The process of my app is quite simple: 

1. I connect to the device
2. I request its services
3. I write/read some characteristics contains in the device (and also subscribe to some) in order to populate my "first view" of the device
4. Then, for my app, I say that the device is connected

Cause of the device itself, sometimes the reading of one characteristics failed: I have set a timeout of 5s on the reading, and the device never answered.

In that case, I restart the process: disconnect the device, connect to the device, try to write. But I think because of the previous failed reading try (on the previous connection), an exception is raised on the first writing (of this new connection):

```
Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "GATT exception from MAC='XX:XX:XX:XX:XX:XX', status 133 (GATT_ERROR), type BleGattOperation{description='CHARACTERISTIC_WRITE'}. (Look up status 0x85 here https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h)")
```

No problem: I disconnect the device and try to connect to the device again but now (and I think because of the exception) all my attempts to connect to the device fails. I can try to restart the device, anything... always it fails. The only thing I can do it's to restart the app.

After many tries, I found a temporary solution which is working, what I do:

When the exception is raised, I deinitialize the `FlutterReactiveBle` lib and then initiliaze it again. In that case, my next attempts to connect to the device succeeds.

**Expected behavior**

Not to have to reinitialize the lib. 

But if the problem is linked to the android BLE stack (or samsung low api) and you can't do a lot of things about it. I will prefer to manage the reinitialization myself like I do now: because if a scan was in progress, I will restart it. But in that case, I will be nice to add something in the documentation to explain that the case may happens.

**Smartphone / tablet**
 - Samsung Galaxy S10+ (Android 11)

**Peripheral device**
- A specific device

werediver, 2021-06-13:
Is this issue reproducible with some other Android device and/or other Android version?

Is it reproducible on iOS?

remonh87, 2021-06-14:
Also an addition to the questions above can you share me the code you use for connecting to a device? Especially in case you use the autoconnect it can sometimes yield to issues when the Ble stack goes bananas.

benoit-rolandeau-act, 2021-06-15:
Hello everyone,

> Is this issue reproducible with some other Android device and/or other Android version?
> 
> Is it reproducible on iOS?

For now, the app is not built for iOS ; therefore I don't know. I tried on a Nokia 7.2 (android one) : 
The same scheme appears (with the exception) but I succeed to connect without reinitializing the BLE library. 

My client also tried on a Samsung S20 and the problem appears. 

I read on internet that there are some problems with the Samsung BLE stacks and ST module (we use a ST), to connect.

> Also an addition to the questions above can you share me the code you use for connecting to a device? Especially in case you use the autoconnect it can sometimes yield to issues when the Ble stack goes bananas.

I also noticed that the autoconnect may have some problems, especially with my S10+ phone, therefore I don't use it. And I manage the reconnection myself within a while loop. 

```
Future<bool> _manageLowLevelConnection(BleDevice device) async {
    final elapsedTime = Stopwatch();
    var leftDuration = ble_constants.connectTimeout;
    DeviceConnectionState state;

    final calculateLeftDuration = () {
      leftDuration = ble_constants.connectTimeout - elapsedTime.elapsed;
    };

    elapsedTime.start();

    while (
        state != DeviceConnectionState.connected && !leftDuration.isNegative) {
      // Listen for the device state in order to know if the connection has
      // succeeded or not
      final completer = Completer();
      final connectionCallback = (DeviceConnectionState connState) {
        state = connState;
        if (connState == DeviceConnectionState.connected ||
            connState == DeviceConnectionState.disconnected) {
          completer.complete();
        }
      };

      final connSub = device.connectionStateStream.listen(connectionCallback);

      Stream<ConnectionStateUpdate> connectionStream;
      AppLogger().d('Try to connect to device: ${device.id}');
      try {
        connectionStream = _flutterBle.connectToDevice(
          id: device.id,
          connectionTimeout: ble_constants.lowLevelConnectTimeout,
        );
      } catch (e) {
        AppLogger().w('The connection of the device: ${device.id}, failed: $e');
      }

      if (connectionStream == null) {
        // The connection fails but we will retry
        await connSub.cancel();
        continue;
      }

      await device.setLibConnectionStream(connectionStream);

      if (device.connectionState == DeviceConnectionState.connected) {
        state = DeviceConnectionState.connected;
        // Device is connected, leave the loop
        await connSub.cancel();
        continue;
      }

      // Calculate the left duration
      calculateLeftDuration();

      try {
        // Wait to receive a connected or disconnected event
        await completer.future.timeout(ble_constants.lowLevelConnectTimeout);
      } catch (error) {
        AppLogger()
            .w("The device ${device.id} isn't connected: $state, and the "
                'timeout raised');
        state = DeviceConnectionState.disconnected;
      }

      if (device.connectionState != DeviceConnectionState.connected) {
        // If the device isn't connected, we force the disconnection
        await device.disconnect();
      }

      await connSub.cancel();

      // Calculate the left duration
      calculateLeftDuration();
    }

    return (state == DeviceConnectionState.connected);
  }
```

From my point of view, the problem doesn't appear on every android phone and not every time. I think when the low BLE stack goes bananas, it will be hard for you to know that a problem occurred.

benoit-rolandeau-act, 2021-06-15:
If I need a while loop here, it's because the app fails to connect multiple times (with the sent of a disconnected state) before succeeds.

remonh87, 2021-06-17:
It is really complicated for me to debug this since it is very specific. I am planning to work on improving the android side the coming weeks (also because android 12 is coming), however it looks some Samsung specific issue (we had quite some of them in the past as well).  I will leave it open since it make sense to add this work around to the docs.

LiangHuangBC, 2021-06-25:
Seems like this is relating to #301

closed 2021-10-20 by remonh87