PhilipsHue/flutter_reactive_ble #288: Cannot build with Xcode13 / IOS 15

DevNico, 2021-06-09:
Multiple errors like the ones below result in a failed build.

ios/Pods/MultiplatformBleAdapter/iOS/RxBluetoothKit/RestoredState.swift:68:105: error: value of optional type 'CBPeripheral?' must be unwrapped to a value of type 'CBPeripheral'
ios/Pods/MultiplatformBleAdapter/iOS/RxBluetoothKit/RxCBCharacteristic.swift:59:52: error: value of optional type 'CBService?' must be unwrapped to a value of type 'CBService'
ios/Pods/MultiplatformBleAdapter/iOS/RxBluetoothKit/RxCBDescriptor.swift:43:62: error: value of optional type 'CBCharacteristic?' must be unwrapped to a value of type 'CBCharacteristic'

remonh87, 2021-06-10:
Thanks we need to incorporate support for the latest operating systems.

DevNico, 2021-06-10:
Whats your priority for this issue / any timeline for when this might get fixed?

It looks like this is an issue in the underlying library used is no longer maintained since Polidea got acquired. Will there be an effort to maintain Polideas libraries or is a switch to corebluetooth for ios on the horizon?

werediver, 2021-06-10:
Polidea's [RxAndroidBlet](https://github.com/Polidea/RxAndroidBle/) is an Android library. flutter_reactive_ble is using CoreBluetooth on iOS.

closed 2021-10-18 by DevNico