PhilipsHue/flutter_reactive_ble #289: App fails to build on Android with cryptic error code (3.1.1+1)

leotoulet, 2021-06-10:
When trying to build the app on an Android device, the build fails with the following error:

```
FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':flutter_reactive_ble'.
> com.android.build.gradle.internal.api.TestVariantImpl_Decorated cannot be cast to com.android.build.gradle.internal.core.InternalBaseVariant

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 6s
```

Flutter doctor -v printout:
```
[√] Flutter (Channel stable, 2.0.3, on Microsoft Windows [Version 10.0.19042.985], locale en-US)
    • Flutter version 2.0.3 at C:\sharing\SDKs\flutter_windows_1.22.6-stable\flutter
    • Framework revision 4d7946a68d (3 months ago), 2021-03-18 17:24:33 -0700
    • Engine revision 3459eb2436
    • Dart version 2.12.2

[√] Android toolchain - develop for Android devices (Android SDK version 30.0.2)
    • Android SDK at C:\Users\******\******\Local\Android\sdk
    • Platform android-30, build-tools 30.0.2
    • Java binary at: C:\Program Files\Android\Android Studio\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b01)
    • All Android licenses accepted.

[√] Chrome - develop for the web
    • Chrome at C:\Program Files\Google\Chrome\Application\chrome.exe

[√] Android Studio (version 4.1.0)
    • Android Studio at C:\Program Files\Android\Android Studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b01)

[√] Connected device (3 available)
    • moto g 7 power (mobile) • ********* • android-arm64  • Android 10 (API 29)
    • Chrome (web)            • chrome     • web-javascript • Google Chrome 90.0.4430.212
    • Edge (web)              • edge       • web-javascript • Microsoft Edge 91.0.864.41

• No issues found!
```

Also tested with JDK 11 with same result.

**Temporary fix:** Downgrade to version 3.0.0 (using `flutter_reactive_ble: '3.0.0'` in pubpsec.yaml), run flutter clean in project, delete all flutter reactive ble folders from pub cache, delete pubspec.lock, and build again (not all steps might be necessary)

Has anyone had this problem?

remonh87, 2021-06-10:
You need to update the gradle wrapper version of your app to `5.6.1` and also make sure to use later version of gradle tools `classpath 'com.android.tools.build:gradle:4.2.1'`. 

Also I noticed you are using a bit older version of flutter so maybe consider running `flutter upgrade` as well (however this is not related to fixing this issue).

leotoulet, 2021-06-10:
Thanks for you answer! closing issue

closed 2021-06-10 by leotoulet