PhilipsHue/flutter_reactive_ble #291: Write to characteristic fails with exception

bosiden, 2021-06-11:
**Describe the bug**
Attempting to write to a write enabled characteristic always returns an exception stating that it is not write enabled.
Using other libraries and BLE test apps I can write successfully without issue.

**To Reproduce**
Steps to reproduce the behavior:
Using write with or without response always returns "not writable" exception.
 I tried to set the characteristic to write only and to write/read without avail.

- [ v] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: iPhone X
 - OS: iOS 14

**Exception**

flutter: #0      Result.dematerialize.<anonymous closure>
package:flutter_reactive_ble/…/model/result.dart:22
#1      Result.iif
package:flutter_reactive_ble/…/model/result.dart:34
#2      Result.dematerialize
package:flutter_reactive_ble/…/model/result.dart:15
#3      ConnectedDeviceOperationImpl.writeCharacteristicWithoutResponse.<anonymous closure>
package:flutter_reactive_ble/src/connected_device_operation.dart:73
#4      _rootRunUnary (dart:async/zone.dart:1362:47)
#5      _CustomZone.runUnary (dart:async/zone.dart:1265:19)
#6      _FutureListener.handleValue (dart:async/future_impl.dart:152:18)
#7      Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:704:45)
#8      Future._propagateToListeners (dart:async/future_impl.dart:733:32)
#9      Future._completeWithValue (dart:async/future_impl.dart:539:5)
#10     _completeOnAsyncReturn (dart:async-patch/async_patch.dart:254:13)
#11     MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart)
package:flutter/…/services/platform_channel.dart:1

[VERBOSE-2:ui_dart_state.cc(186)] Unhandled Exception: Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "The characteristic 44A78CE3-E653-4827-8AAD-8A08268056BA of the service 0EF881BC-6EC4-11EA-BC55-0242AC130003 of the peripheral 57FD66C2-0FE4-78CA-D719-F13AC93CECF7 is not writable")

bosiden, 2021-06-11:
Some more information:
Running on Android works without issues, write to characteristic works fine on Android, fails on iOS.
Reading on iOS works fine.

bosiden, 2021-06-12:
Tested on another iPhone 7 running iOS 14.4 and write fails as well with the same exception error.

bosiden, 2021-06-13:
Turns out I had wrong characteristic parameters, they worked fine on Android, but iOS was picky.

closed 2021-06-13 by bosiden

rafaldziuryk, 2021-06-25:
Hi,

@bosiden, what was your problem? I have the same issue :(

JKornberg, 2021-07-06:
Also having this issue where it worked on Android, what characteristic paramters are you talking about @bosiden

rafaldziuryk, 2021-07-12:
@JKornberg - try to change `writeCharacteristicWithResponse` to `writeCharacteristicWithoutResponse` or vice versa