PhilipsHue/flutter_reactive_ble #292: A way of getting RSSI of connected device

bartlomiejszm, 2021-06-14:
**Is your feature request related to a problem? (please describe)**
I do have a problem where i have to have signal strength fo my bluetooth device, the device stops advertising after being connected, so i am unable to read the RSSI.

**Describe the solution you'd like**
it would be nice to have a possibility to listen to a stream of changing RSSI

**Describe alternatives you've considered**
i have considered scanning constantly but my device is not advertising after being connected.

**Additional context**
well thats pretty much that is to it

bartlomiejszm, 2021-06-14:
On a side note, i have noticed that another package https://pub.dev/packages/flutter_blue does have this feature, at least they state it has in their issues, but it doesnt at all fit into my logic, i need the reactiveness this package provides.

remonh87, 2021-06-14:
Thank you for interest in our library.  I think it is a good feature that would be very nice to have. As you know both Android12 and iOS15 are upcoming and this will be our focus on the short term. I will label this as enhancement. 

Btw if you are up to it feel free to create a pr yourself :)

Fasuh, 2021-06-14:
I might try to do this, but i am also a little busy right now, we'll see

remonh87, 2021-06-14:
understandable make sure to assign the issue if you are working on it.

cjcdev, 2021-11-29:
This would be really helpful to have.  Some apps can connect to multiple devices, but would like to know the 'nearest' based on RSSI.

raphael-bmec-co, 2022-01-04:
Hi @Taym95

No expectations, but do you have a timeline on this? If not, our team may have to implement it.

Thanks.

TNorbury, 2022-08-03:
I'm working on this currently, will have a PR up later today or tomorrow

ziperjohn, 2023-07-19:
Hello

what is the status of this feature are you planning to add it to the next version of the plugin? I think this feature would take this plugin to the next level.

Taym95, 2024-01-25:
available in next release

closed 2024-01-25 by Taym95

raphael-bmec-co, 2024-04-16:
Hello

Is there any documentation and/or an example on how to access this API? I've had a look but I'm struggling to find anything.

Taym95, 2024-04-16:
you need `v 5.3.1 ` and call `readRssi` function

raphael-bmec-co, 2024-04-16:
Perfect thanks. Simple enough 🤦 

I'm not clear on the underlying implementation but is this value available as a stream? If not, is there an underlying stream at the platform layer and an appropriate maximum rate at which we should call `readRssi`?

Basically, we have an RSSI sensitive application and we'd like to monitor this value but I don't want to be calling it 10x per second if 1) it has any BLE overhead - i.e. it actually queries the server or 2) it is only actually updated every 5 seconds for example.

Taym95, 2024-04-16:
> Perfect thanks. Simple enough 🤦
>
> I'm not clear on the underlying implementation but is this value available as a stream? If not, is there an underlying stream at the platform layer and an [...]

there is not stream, it is just future but you can keep recalling it and check the rate limit