PhilipsHue/flutter_reactive_ble #293: A way to handle specific GATT errors

jobfeikens, 2021-06-16:
I have a write operation that returns the following error: 
```
I/flutter ( 2244): Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "GATT exception from MAC='XX:XX:XX:XX:XX:XX', status 3 (GATT_WRITE_NOT_PERMIT), type BleGattOperation{description='CHARACTERISTIC_WRITE'}. (Look up status 0x03 here https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h)")
```
It would be nice if it were possible to extract the status code (3) from this exception. This is currently only possible if you convert the exception to a string and then somehow extract the status code from that string.

remonh87, 2021-06-16:
This is a good request and maybe we can add some granularity in error codes indeed. I would expect iOS also has a specific error or code that we can map in case writee is not permitted. @werediver  can you confirm?

werediver, 2021-06-17:
On iOS we essentially get an [NSError](https://developer.apple.com/documentation/foundation/nserror/1409165-code?language=objc) instance in case of a failure. It has the following fields:

- a text `domain`
- an integer `code`
- and a dictionary `userInfo`

It should be simple to pass through the domain and the code, and I'd suggest to ignore `userInfo` for the first iteration (not clear, if it would be useful and in which cases).

rohitsangwan01, 2021-12-18:
@werediver , till getting an official update , you can try out this silly workaround

```dart
  if (connectionState.failure != null) {
                  String ErrorMessage = connectionState.failure!.code.toString() + " : " +  connectionState.failure!.message;

                  ///First Check if Error Contains Error Status Code
                  if (connectionState.failure!.message.contains('status')) {
                    String ErrorMsg = connectionState.failure!.message;
                    var ErrorArray = ErrorMsg.split(" ");
                    var ErrorCodeIndex = ErrorArray.indexOf('status') + 1;
                    String ErrorCode = ErrorArray[ErrorCodeIndex];
                    Get.log('Status Code : $ErrorCode');
                   //now parse status code to Bluetooth Error Message by BleErrorConvertor (check below)
                    if (BleErrorConvertor(ErrorCode) != 'NO') {
                      ErrorMessage = BleErrorConvertor(ErrorCode);
                    }
                  }
             
                 //Show Error popu With Error Messgae
                 // Ui.Error(ErrorMessage, duration: Duration(seconds: 2));
                } else {
                //  Ui.Error('Disconnected From ' + device.name);
                }
```
BleErrorConvertor method to return error message from status code 

```dart
BleErrorConvertor(String value) {
  int? data = int.tryParse(value);

  if (data == null) return 'NO';

  switch (data) {
    case 10:
      return 'Connection Timeout';
    case 19:
      return 'Remote User Terminated Connection';
    case 21:
      return 'Remote Device Terminated Connection';
    case 22:
      return 'Connection Terminated By Local Host';
    case 62:
      return 'Connection Failed To Be Established / Synchronization Timeout';
    default:
      return 'NO';
  }
}

```