PhilipsHue/flutter_reactive_ble #297: Unable to build flutter_reactive_ble

adailydose, 2021-06-23:
**Describe the bug**
During the build process, I keep getting this error:
```
* What went wrong:
A problem occurred configuring project ':flutter_reactive_ble'.
> class com.android.build.gradle.internal.api.TestVariantImpl_Decorated cannot be cast to class com.android.build.gradle.internal.core.InternalBaseVariant (com.android.build.gradle.internal.api.TestVariantImpl_Decorated is in unnamed module of loader org.gradle.internal.classloader.VisitableURLClassLoader @59aafbb5; com.android.build.gradle.internal.core.InternalBaseVariant is in unnamed module of loader org.gradle.internal.classloader.VisitableURLClassLoader @5a8d0efa)
```

**To Reproduce**
Built using Android Studio 4.2.1
[√] Flutter (Channel stable, 2.2.2, on Microsoft Windows [Version 10.0.19042.1052], locale en-US)
    • Flutter version 2.2.2 at C:\src\flutter
    • Framework revision d79295af24 (12 days ago), 2021-06-11 08:56:01 -0700
    • Engine revision 91c9fc8fe0
    • Dart version 2.13.3

**Expected behavior**
Build without error.

remonh87, 2021-06-23:
This happens most of the time when you are using an older version of either android tools or gradle wrapper.
Make sure you use the following:
`com.android.tools.build:gradle:4.2.1`
gradle wrapper 5.6.1 or above

adailydose, 2021-06-23:
Thanks for the tip!

closed 2021-06-23 by adailydose