PhilipsHue/flutter_reactive_ble #298: Auto reconnect is not working, when shutdown bt on target device and turn it on again

LiangHuangBC, 2021-06-23:
Reproduce steps:
1. My android pad(Amazon Fire 8) connected to a BT device. On Flutter run example app.
2. Shut BT device down. 
logs:
`D/BluetoothGatt(31806): onClientConnectionState() - status=8 clientIf=7 device=3C:A5:51:84:7D:01
D/BluetoothManager(31806): getConnectionState()
D/BluetoothManager(31806): getConnectedDevices
I/flutter (31806): ======ConnectionStateUpdate(deviceId: 3C:A5:51:84:7D:01, connectionState: DeviceConnectionState.disconnected, failure: GenericFailure<ConnectionError>(code: ConnectionError.failedToConnect, message: "Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 8 (GATT_INSUF_AUTHORIZATION or GATT_CONN_TIMEOUT)"))========
D/BluetoothGatt(31806): close()
D/BluetoothGatt(31806): unregisterApp() - mClientIf=7
W/libEGL  (31806): EGLNativeWindowType 0xd4547008 disconnect failed
W/libEGL  (31806): EGLNativeWindowType 0xd45ba808 disconnect failed`

3. Turn BT device on again.  Example app on my pad has no reaction.

**Expected behavior**
Auto connect to BT device when BT is available.

Tried default connect method and "connectToAdvertisingDevice" method, both are not working.
Tried with or without "connectionTimeout" parameter, both are not working.

 _connection = _ble.connectToAdvertisingDevice(
        id: deviceId,
        prescanDuration: const Duration(seconds: 3),
        // connectionTimeout: const Duration(seconds: 2),
        withServices: [
          serviceUuid,
          charUuid,
        ]).listen(
      (update) {
        print("======$update========");
        _logMessage(
            'ConnectionState for device $deviceId : ${update.connectionState}');
        _deviceConnectionController.add(update);
      },
      onError: (Object e) =>
          _logMessage('Connecting to device $deviceId resulted in error $e'),
    );

LiangHuangBC, 2021-06-24:
Passed it around by trying to reconnect to lost device in loop manually.

remonh87, 2021-06-25:
For some phones and  peripheral combinations auto connect is a good option but for others it should be avoided. This is unfortunately something inside the bluetooth stack in Android or of the device itself. At Hue we also do not use auto connect.

Can you try to connect the first time without autoconnect (by specifying a timeout) and check if the issue still occurs?

LiangHuangBC, 2021-06-25:
Solved here #300

closed 2021-06-25 by LiangHuangBC