PhilipsHue/flutter_reactive_ble #299: App crashes on iOS when trying to connect to device.

kutlayhanli-maxim, 2021-06-24:
**Describe the bug**
Devices are found using scanForDevices but iOS can't connect to device.

**To Reproduce**

***info.plist:***

	<key>NSBluetoothAlwaysUsageDescription</key>
	<string>The app uses bluetooth to find, connect and transfer data between different devices</string>
	<key>NSBluetoothPeripheralUsageDescription</key>
	<string>The app uses bluetooth to find, connect and transfer data between different devices</string>
	<key>NSLocalNetworkUsageDescription</key>
	<string>The app uses bluetooth to find, connect and transfer data between different devices</string>

***podfile:***

post_install do |installer|
  installer.pods_project.targets.each do |target|
    flutter_additional_ios_build_settings(target)
    target.build_configurations.each do |config|

      config.build_settings['GCC_PREPROCESSOR_DEFINITIONS'] ||= [
        '$(inherited)',

        ## dart: PermissionGroup.bluetooth
        'PERMISSION_BLUETOOTH=1',
      ]

    end
  end
end

***Dart source:***

flutterReactiveBle.connectToDevice(
  id: devices[index].id,
  connectionTimeout: const Duration(seconds: 2),
);

*output:*

*** Assertion failure in -[CBUUID initWithData:], CBUUID.m:195
*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Data {length = 0, bytes = 0x} does not represent a valid UUID'
*** First throw call stack:
(0x196656754 0x1ab11d7a8 0x1965583f8 0x19792c714 0x1afe96bf8 0x1afe96ddc 0x1afe86c78 0x1afe95cb0 0x1afe94bf8 0x1afe94d04 0x1afe9136c 0x1afe86150 0x1afeadf78 0x1afea1874 0x196241a84 0x19624381c 0x19624b004 0x19624bc34 0x196251b5c 0x1965d0398 0x1965ca270 0x1965c9360 0x1adc07734 0x199044584 0x199049df4 0x102207040 0x196285cf8)
libc++abi: terminating with uncaught exception of type NSException
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00000001c476d334 libsystem_kernel.dylib`__pthread_kill + 8
libsystem_kernel.dylib`__pthread_kill:
->  0x1c476d334 <+8>:  b.lo   0x1c476d354               ; <+40>
    0x1c476d338 <+12>: pacibsp 
    0x1c476d33c <+16>: stp    x29, x30, [sp, #-0x10]!
    0x1c476d340 <+20>: mov    x29, sp
Target 0: (Runner) stopped.
Lost connection to device.

**Expected behavior**
It works as expected on Samsung Galaxy A11 - Android 11.
On Android device is connected and I can read-write characteristics as expected.

**Smartphone / tablet**
 - Device: iPhone XR
 - OS: iOS 14.6

**Peripheral device**
- Custom prototype device

**Additional context**
-

werediver, 2021-06-24:
You missed an important part of the issue template: trying to do whatever you're having issue with in a generic BLE scanner app like "nRF Connect". First of all I'd suspect either a peripheral issue, or invalid parameters passed to the library.

kutlayhanli-maxim, 2021-06-24:
> You missed an important part of the issue template: trying to do whatever you're having issue with in a generic BLE scanner app like "nRF Connect". First of all I'd suspect either a peripheral issue, or invalid parameters passed to the library.

Great advice. NRF Connect has the same behavior, works on Android crashes on iOS. The issue is definitely with our prototype device. We narrowed down the issue: app crashes when # of characteristics exceeds a limit. Something goes wrong in prototype device, probably memory management. This occurs when iOS device tries to read available services and characteristics upon connecting.

For rapid prototyping **is there a quick way to disable polling for available services and characteristics upon connecting?** In our specific application mobile app already knows what the IOT device has as services and characteristics and we handled errors for read/write.

@mehmetariman
@karaanil

werediver, 2021-06-25:
It's usually called service and characteristic discovery and you can avoid discovering _all_ services and characteristic by specifying which one you are _actually_ going to use in your application. Please refer to the documentation and/or implementation of [FlutterReactiveBle.connectToDevice(id:, servicesWithCharacteristicsToDiscover:, connectionTimeout:)](https://github.com/PhilipsHue/flutter_reactive_ble/blob/c5acb31586353e5ec8eda48bbd503c073d707bbd/lib/src/reactive_ble.dart#L264), pay attention to `servicesWithCharacteristicsToDiscover:` parameter (it's a dictionary of format `[service1: [char11, char12], service2: [char21, char22]]`). It only has effect on iOS. Don't forget to keep this dictionary up to date, otherwise you'll get errors (when trying to use undiscovered services or characteristics).

iBog, 2022-02-03:
Same issue:
2022-02-03 12:49:48.989080+0300 Runner[788:182307] flutter: BLE device found!
Warning! No event channel set up to report a connection update
reactive_ble_mobile/ConnectTaskController.swift:16: Assertion failed
2022-02-03 12:49:48.991311+0300 Runner[788:182053]
 reactive_ble_mobile/ConnectTaskController.swift:16: Assertion failed
(lldb) 

[image]

[image]

egops, 2022-03-11:
@iBog same issue