PhilipsHue/flutter_reactive_ble #300: UndeliverableException

LiangHuangBC, 2021-06-24:
**Describe the bug**
Turn bluetooth down in android. Connect to device, and got an UndeliverableException. Even added dependency in build.gradle
`implementation "com.polidea.rxandroidble2:rxandroidble:1.12.1"`

Relating to bug #298 

**Expected behavior**
Flutter side catch this exception.

Logs:

E/AndroidRuntime(32568): io.reactivex.exceptions.UndeliverableException: The exception could not be delivered to the consumer because it has already canceled/disposed the flow or the exception has nowhere to go to begin with. Further reading: https://github.com/ReactiveX/RxJava/wiki/What's-different-in-2.0#error-handling | com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status -1 (UNKNOWN)
E/AndroidRuntime(32568): 	at io.reactivex.plugins.RxJavaPlugins.onError(RxJavaPlugins.java:367)
E/AndroidRuntime(32568): 	at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver.onError(ObservableUnsubscribeOn.java:67)
E/AndroidRuntime(32568): 	at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.tryOnError(ObservableCreate.java:84)
E/AndroidRuntime(32568): 	at com.polidea.rxandroidble2.internal.util.DisposableUtil$3.onError(DisposableUtil.java:59)
E/AndroidRuntime(32568): 	at io.reactivex.internal.operators.single.SingleDoFinally$DoFinallyObserver.onError(SingleDoFinally.java:79)
E/AndroidRuntime(32568): 	at io.reactivex.internal.operators.single.SingleTimeout$TimeoutMainObserver.onError(SingleTimeout.java:150)
E/AndroidRuntime(32568): 	at io.reactivex.internal.operators.single.SingleCreate$Emitter.tryOnError(SingleCreate.java:94)
E/AndroidRuntime(32568): 	at com.polidea.rxandroidble2.internal.util.DisposableUtil$1.onError(DisposableUtil.java:23)
E/AndroidRuntime(32568): 	at io.reactivex.internal.operators.flowable.FlowableElementAtSingle$ElementAtSubscriber.onError(FlowableElementAtSingle.java:101)
E/AndroidRuntime(32568): 	at io.reactivex.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.checkTerminate(FlowableFlatMap.java:572)
E/AndroidRuntime(32568): 	at io.reactivex.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.drainLoop(FlowableFlatMap.java:379)
E/AndroidRuntime(32568): 	at io.reactivex.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.drain(FlowableFlatMap.java:371)
E/AndroidRuntime(32568): 	at io.reactivex.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.innerError(FlowableFlatMap.java:611)

LiangHuangBC, 2021-06-25:
Read #88   but my flutter use kotlin, I only have a MainActivity.kt, no idea of android native.... any detail guid?

remonh87, 2021-06-25:
We have a work around for the Undeliverable exception documented here: https://github.com/PhilipsHue/flutter_reactive_ble#faq . However use this workaround only if you know for certain surpressing this behavior is correct. In order to further diagnose the issue can you send me the code snippet for connecting? Also does this problem occur on our example app?

LiangHuangBC, 2021-06-25:
Solved by adding this in MainActivity.kt 
`class MainActivity: FlutterActivity() {
    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        RxJavaPlugins.setErrorHandler { throwable ->
            if (throwable is UndeliverableException && throwable.cause is BleException){
                return@setErrorHandler
            }
            throw RuntimeException("Unexpected Throwable in RxJavaPlugins error handler", throwable)
        }
    }
}`

LiangHuangBC, 2021-06-25:
I did some trick in my code, when a BEL device is turned down, my code keep connecting it in loop until success. Looks like there is no UndeliverableException happened now. But got another error, if I cancel a stream subscription to disconnect to a "connecting" device(ble device is turned down) . I cannot connect to any device anymore. I have to restart my app to connect again.

LiangHuangBC, 2021-06-25:
`connect(DiscoveredDevice device) async {
    if (btState.value != BleStatus.ready) return;
    this.device.value = device;

    await _connSub?.cancel();
    await _subChar?.cancel();
    _connSub = flutterReactiveBle
        .connectToDevice(
      id: device.id,
      servicesWithCharacteristicsToDiscover: {
        svcId: [
          charId,
        ]
      },
      connectionTimeout: const Duration(seconds: 10),
    )
        .listen((connectionStateUpdate) {
      // Handle connection state updates
      logger.i(connectionStateUpdate);
      deviceState.value = connectionStateUpdate.connectionState;
      if (deviceState.value == DeviceConnectionState.connected) {
        updateDevices(device);
        readData(device);
        saveDeviceLocal(device.id, device.name);
        flutterReactiveBle.requestConnectionPriority(
            deviceId: device.id, priority: ConnectionPriority.highPerformance);
      } else if (deviceState.value == DeviceConnectionState.disconnected) {
        // disconnected due to lost signal or shuting bt down
        // try reconnect
        connect(device);
      }
    }, onError: (Object error) {
      logger.e(error);
    });
  }`

LiangHuangBC, 2021-06-25:
Close this and add #301

closed 2021-06-25 by LiangHuangBC