PhilipsHue/flutter_reactive_ble #301: Disconnect while connecting, can not connect to any device anymore

LiangHuangBC, 2021-06-25:
**Describe the bug**
Disconnect while connecting, can not connect to any device anymore

**To Reproduce**
Steps to reproduce the behavior:
1. Run Example app
2. Scan and found two device. Connect to anyone works well.
3. Turn deviceA's power down. 
4. Connect to Device A, it shows connecting.
5. Return to scan list, connect to DeviceB, it show connecting forever. 
6. Now you cannot connect to any device anymore.  Unless you restart app, or turn Device A on and connect to device A will be succeed, and everything works.

**Expected behavior**
After disconnecting from connecting device, it can connect to other device.

**Smartphone / tablet**
Run example app on Amazon Fire 8.

LiangHuangBC, 2021-06-25:
tried deinitialize(), no luck

benoit-rolandeau-act, 2021-06-28:
From my experience, the auto connection doesn't work well with some smartphones and may cause this problem. Do you use the auto connection?

LiangHuangBC, 2021-06-28:
I turned auto to false, same issue.

benoit-rolandeau-act, 2021-06-28:
Did you receive a "Disconnected" event when connecting to DeviceB (when you say that you wait forever)?

remonh87, 2021-06-28:
@LiangHuangBC  do you use the `connectToDevice ` method  or the `connectToAdvertisingDevice ` ?  The last one is more safe since we will scan for the availability of the device before we connect to it. If you use the connectToDevice and the device is powered off you run into the issue that Android will attempt to connect forever . This is something in the OS and we cannot fix this. I suggest to use `connectToAdvertisingDevice` in that case.

remonh87, 2021-07-04:
Closing for now feel free to reopen it in case you have additional questions

closed 2021-07-04 by remonh87

LiangHuangBC, 2021-07-23:
Actually you can fix it. I passed it around. When disconnect, wait for connecting timeout. Then disconnecting works and all other connecting work.

yuriboeira11tx, 2021-09-28:
How to set ```autoConnect``` to false?