PhilipsHue/flutter_reactive_ble #302: Why manufacturer data's first two byte is always in reverse order?

keval-devani, 2021-06-25:
**Describe the bug**
When scanning BLE device only first two bytes in manufacturerData order is reverse. 

**To Reproduce**
Steps to reproduce the behavior:
1. set manufacturer Data in nrf connect for mobile 
- Company Identifier : 0059 
- Data : 5403202121.....
2. When Scanning  in result device.manufacturerData give me data like this **[89, 0, 84, 3, 32, 33, 33]** that means **59 00** 

**Expected behavior**
it should not reverse first two bytes in manufacturerData. 

- [ ] I tried doing the same with a general BLE scanner application nrf connect and also try with flutter_ble lib that give me **00 59** in result

**Smartphone / tablet**
 - Device: Moto moto one 
 - OS:  Android 10
 - Package version: flutter_reactive_ble: ^3.1.1+1

**Peripheral device**
- Vendor, model: nrf connect app for mobile (Poco F1)
- Does it run a custom firmware / software: no

**Additional context**
-

jobfeikens, 2021-06-28:
Hi, its not in reverse.

From v4.2 of the spec:
```
2.4 Profile Fundamentals

    ...

    • Multi-octet fields within the GATT Profile shall be sent least significant octet first (little endian).
```

werediver, 2021-06-28:
Thank you for looking into this, @jobfeikens.

I suppose, the above answer resolves the original issue.

closed 2021-06-28 by werediver

keval-devani, 2021-06-30:
But, what about v5? Is there is way to differentiate that it is coming from v5 or v4.2?

werediver, 2021-07-03:
The same is true for BLE v5.0, see the Core spec p. 2225.

No, there is no (direct) way of determining the version of the BLE that your device supports.