PhilipsHue/flutter_reactive_ble #304: Long IDs with padding zeroes are shortened on iOS but not on Android.

dvigara, 2021-06-28:
**Describe the bug**
When discovering services in iOS with the sample app, the services and characteristics with IDs starting with several zeros have their IDs shortened:

[image]

In android the id of this service is read without issues: "serviceId: 0000180a-0000-1000-8000-00805f9b34fb": 

[image]

**To Reproduce**
1. With the sample app running on iOS, connect to a device with a long service or characteristc's ID and with starting zeros
2. Discover services or simply read the advertised services

**Expected behavior**
IDs are read exactly the same as in Android

**Smartphone**
 - Device: iPhone 8 plus
 - OS: iOS 14.6
 - Package version: 3.1.1

**Peripheral device**
- Vendor, model: CUSTOM
- Does it run a custom firmware: yes

remonh87, 2021-07-04:
@werediver  can you take a look at this?

werediver, 2021-07-04:
Hi, @remonh87.

On iOS the library behaves transparently and preserves the IDs as they are reported by the system BLE stack (BLE supports short 16- and 32-bit IDs; never saw a 32-bit ID in practice, though).

On Android the native part extends the sort IDs to the full 128-bit length for some reason (see [UuidConverter.kt](https://github.com/PhilipsHue/flutter_reactive_ble/blob/bc030414a540c92b00dc363891c8f1013eaf807e/android/src/main/kotlin/com/signify/hue/flutterreactiveble/converters/UuidConverter.kt)).

The discrepancy is unfortunate, but I am yet to be convinced this implicit ID expansion is the right thing to do (that's not what the system BLE stacks do, on both  platforms).

I would rather suggest considering removing the implicit ID expansion in the Android part, unless there is a strong reason for that (which we should document then and think whether the discrepancy is acceptable).

werediver, 2021-07-09:
@remonh87 This behavior also causes issues in Hue BT app after switching to the latest (null-safe) version of the library (so it wasn't in the library always).

remonh87, 2021-07-09:
hmm maybe this is the issue: https://github.com/PhilipsHue/flutter_reactive_ble/pull/217

werediver, 2021-07-09:
@remonh87 Ehm, no. As I mentioned above, this is the issue: [UuidConverter.kt](https://github.com/PhilipsHue/flutter_reactive_ble/blob/bc030414a540c92b00dc363891c8f1013eaf807e/android/src/main/kotlin/com/signify/hue/flutterreactiveble/converters/UuidConverter.kt).

remonh87, 2021-07-12:
@werediver  
Ok I did some research and came to the following conclusion:
- this class is not changed since november 2019  so there is no regression.
-  Android only supports 128bit variants so when we need to construct the UUID from the data in protobuf we are forced to show "fill" the remaining space with zeroes. See https://developer.android.com/reference/java/util/UUID .

So I do not see anything I can do about it besides documenting the difference

closed 2021-07-24 by remonh87