PhilipsHue/flutter_reactive_ble #306: Example is missing required files

Lerie82, 2021-07-01:
```
import 'package:flutter_reactive_ble_example/src/ble/ble_device_connector.dart';
import 'package:flutter_reactive_ble_example/src/ble/ble_device_interactor.dart';
import 'package:flutter_reactive_ble_example/src/ble/ble_scanner.dart';
import 'package:flutter_reactive_ble_example/src/ble/ble_status_monitor.dart';
import 'package:flutter_reactive_ble_example/src/ui/ble_status_screen.dart';
import 'package:flutter_reactive_ble_example/src/ui/device_list.dart';
import 'package:provider/provider.dart';

import 'src/ble/ble_logger.dart';
```

I understand they are in the repo but maybe a copy+paste example might be better.

remonh87, 2021-07-01:
can you elaborate on what you mean with example is missing required files? Is it in the readme?  The readme mainly contains small snippets and the example app provides normally a copy paste solution to get started.

werediver, 2021-07-03:
The library provides a substantial functionality, and the example project covers a substantial part of it. The complexity of the example project requires a certain architectural design to make the example digestible.

I recommend you to check out the repository or download it as a zip archive (you can as well download the archived sources from a release page) and open the example project in an IDE, so that you can comfortably explore it.

closed 2021-07-03 by werediver