PhilipsHue/flutter_reactive_ble #307: PluginController.swift Assertion failed / iOS

MazenCoder, 2021-07-04:
flutter_reactive_ble/PluginController.swift:86: Assertion failed Runner[36443:1970108] flutter_reactive_ble/PluginController.swift:86: Assertion failed

JKornberg, 2021-07-06:
I'm having a similar issue:

```
flutter_reactive_ble/PluginController.swift:153: Assertion failed
* thread #1, queue = 'com.apple.main-thread', stop reason = Assertion failed
    frame #0: 0x0000000191969060 libswiftCore.dylib`_swift_runtime_on_report
libswiftCore.dylib`_swift_runtime_on_report:
->  0x191969060 <+0>: ret
libswiftCore.dylib`_swift_reportToDebugger:
    0x191969064 <+0>: b      0x191969060               ; _swift_runtime_on_report
libswiftCore.dylib`_swift_shouldReportFatalErrorsToDebugger:
    0x191969068 <+0>: adrp   x8, 323753
```

However I"m also unable to successfully connect on iOS

JKornberg, 2021-07-06:
I noticed I'm getting my crash when the connected bluetooth device goes to sleep. Here is the code for my connect function:
```
  connectToDevice(DiscoveredDevice device) async {
    try {
      await _connectionStateSubscription.cancel();
    } catch (error) {
      print(error);
    }
    _connectionStateSubscription = ble
        .connectToDevice(
      id: device.id,
      servicesWithCharacteristicsToDiscover: {
        service: [characteristic]
      },
      connectionTimeout: const Duration(seconds: 2),
    )
        .listen((connectionState) {
      print("New connection state: $connectionState");
      if (connectionState.connectionState == DeviceConnectionState.connected) {
        _device.sink.add(device);
        _connectionState.add(connectionState);
        writeResult(label: "Connected", value: "Connected to device");
      } else {
        _device.sink.add(null);
        _connectionState.add(null);
        writeResult(label: "Disconnected", value: "Lost connection to device");
      }
    }, onError: (dynamic error) {
      print("ERROR: " + error.toString());
      _connectionState.add(null);
    });
  }
```

MazenCoder, 2021-07-06:
> I noticed I'm getting my crash when the connected bluetooth device goes to sleep. Here is the code for my connect function:
>
> ```
>   connectToDevice(DiscoveredDevice device) async {
>     try { [...]

I solved it by changing the : assert(false); to print("");
open the xcode and change: -> else {assert(false) return} to: -> else {print(""); return}

werediver, 2021-07-06:
> I solved it by changing the : assert(false); to print("");
> open the xcode and change: -> else {assert(false) return} to: -> else {print(""); return}

This is not a solution in the slightest. The assert is there for a reason. If it triggers (which only happens in debug builds), it means the code is in an invalid. In this case the invalid state may be caused by a overly aggressive (rapid) sequence of API calls (which should never cause issues, ideally; in practice it can, though 😞).

MazenCoder, 2021-07-06:
> > I solved it by changing the : assert(false); to print("");
> > open the xcode and change: -> else {assert(false) return} to: -> else {print(""); return}
> 
> This is not a solution in the slightest. The assert is there for a reason. If it triggers (which only happens in debug builds), it means the code is in an invalid state. In this case the invalid state may be caused by a overly aggressive (rapid) sequence of API calls (which should never cause issues, ideally; in practice it can, though 😞).

I agree with you but this is the solution I took for now
and It works and the data is sent. This is my code:

`class BleDeviceConnector extends ReactiveState<ConnectionStateUpdate> {
  BleDeviceConnector(this._ble);

  final FlutterReactiveBle _ble;
  bool isConnected = false;

  @override
  Stream<ConnectionStateUpdate> get state => _deviceConnectionController.stream;

  final _deviceConnectionController = StreamController<ConnectionStateUpdate>();

  StreamSubscription<ConnectionStateUpdate> _connection;

  Future<void> connect(String deviceId) async {
    if (_connection != null) {
      await _connection.cancel();
    }

    _connection = _ble.connectToDevice(id: deviceId).listen(
          _deviceConnectionController.add,
        );
  }

  Future<bool> connect2(String deviceId) async {
    try {
      print('connect2: $deviceId');
      this.isConnected = false;
      if (_connection != null) {
        await disconnect(deviceId);
        // await _connection.cancel();
      }
      // disconnect(deviceId);
      return await _connect(deviceId);
      // return await Future.delayed(Duration(seconds: 15), () => this.isConnected);

    } catch(e) {
      print('error: $e');
      return false;
    }
  }

  Future<bool> _connect(String deviceId) async {
    try {
      final scanStream = _ble.connectToDevice(id: deviceId);
      await for (ConnectionStateUpdate state  in scanStream) {
        print('state: ${state.connectionState}');
        if (state.connectionState == DeviceConnectionState.connecting) {
          continue;
        } else if (state.connectionState == DeviceConnectionState.connected) {
          _connection = _ble.connectToDevice(id: deviceId).listen(
            _deviceConnectionController.add,
          );
          this.isConnected = true;
          return this.isConnected;
        } else {
          this.isConnected = false;
          return this.isConnected;
        }
      }
      return this.isConnected ?? false;
    } catch(e) {
      print('error: $e');
    }
  }

  Future<bool> connect3(String deviceId) async {
    try {
      if (_connection != null) {
        await _connection.cancel();
      }

      final scanStream = _ble.connectToDevice(id: deviceId);

      await for (ConnectionStateUpdate state  in scanStream) {
        print('state: ${state.connectionState}');
        if (state.connectionState == DeviceConnectionState.connecting) {
          continue;
          // return await Future.delayed(Duration(seconds: 30)).then((value) => false);
        } else if (state.connectionState == DeviceConnectionState.connected) {
          return true;
        } else if (state.connectionState == DeviceConnectionState.disconnected) {
          return false;
        }  else {
          return false;
        }
      }
      await discoverServices(deviceId);
    } catch(e) {
      print('error: $e');
      return false;
    }
  }

  Future<void> disconnect(String deviceId) async {
    print('disconnect: $deviceId');
    if (_connection != null) {
      try {
        await _connection.cancel();
      } on Exception catch (e, _) {
        print("Error disconnecting from a device: $e");
      } finally {
        // Since [_connection] subscription is terminated, the "disconnected" state cannot be received and propagated
        _deviceConnectionController.add(
          ConnectionStateUpdate(
            deviceId: deviceId,
            connectionState: DeviceConnectionState.disconnected,
            failure: null,
          ),
        );
      }
    }
  }

  Future<void> discoverServices(String deviceId) async {
    await _ble.discoverServices(deviceId).then(
          (value) {
            debugPrint('Services discovered: $value');
            log('data: $value');

          },
        );
  }

  Future<void> dispose() async {
    await _deviceConnectionController.close();
  }
  
  Future<void> writeChar({String deviceId, List<int> value}) async {
    final characteristic = QualifiedCharacteristic(serviceId: Uuid.parse('0000ffe0-0000-1000-8000-00805f9b34fb'), characteristicId: Uuid.parse('0000ffe1-0000-1000-8000-00805f9b34fb'), deviceId: deviceId);
    await _ble.writeCharacteristicWithoutResponse(characteristic, value: value);
  }

  Future<bool> _break() => Future.delayed(Duration(seconds: 10)).then((value) => false);
  
}`

lgozalvez, 2021-07-07:
Same issue here :( 

It is also a random issue for us.

ju851zel, 2021-07-22:
Maybe you have to call stopScanning(). See https://github.com/PhilipsHue/flutter_reactive_ble/issues/283#issuecomment-850858995

gonzalolopez-tc, 2021-09-09:
any update about this issue?

saibotma, 2021-09-10:
Also having the same issue.

vincent290587, 2021-09-14:
Also having this issue...

shobhit14gupta, 2021-09-14:
Also having this issue while multiple subscribing to characteristics.

lgozalvez, 2021-09-14:
We all have the same issue...

remonh87, 2021-09-14:
Like previously said by @werediver the assert is there for a reason.  Instead it would be great if you can help us with providing a reproducible example or additional logging where we can triage it better and determine if it is a mistake by our library

vincent290587, 2021-09-19:
I will try to give it a go then ;-)
I am fairly new to (flutter) app dev, so please if you need anything just ask !

Some BLE services send a packet right after connection to set the inital value of the characteristic. This is actually my case.
So the phone should expect an incoming packet very shortly after connecting.

I went in debug mode on my iPhone and did verify that:
- _onCharacteristicValueUpdate_ is called immediately after connecting
- _charactertisticValueStreamHandler_ **onListen** is called after _onCharacteristicValueUpdate_

However the characteristicValueUpdateSink is only set in the later function, and if it is not already set in _onCharacteristicValueUpdate_ , the assert will trigger.
It looks like a ce racondition where the stream should be listened to before the connection is really established.

vincent290587, 2021-09-19:
If you have a nRF52832 dev kit from Nordic I can provide you with a binary

closed 2021-10-01 by remonh87

jayrudd02, 2022-08-06:
Guys, I faced a similar problem.
Using Bluetooth
[image]
...

I am using a mac pro, and what solved it for me was editing the scheme of the app from debug to release within the product tab in Xcode

PR #393 (2021-10-01): ios: Queue up messages
Queue up messages until event-sink has been created
Closes #307, #251, #137, #387
commit fc331afb (2021-10-01): ios: Queue up messages
Queue up messages until event-sink has been created
Closes #307, #251, #137