PhilipsHue/flutter_reactive_ble #308: Question: What does "A scan for a different service is running" mean in a ConnectionStateUpdate

JKornberg, 2021-07-06:
Hi, This package is working pretty well on Android, but for iOS I am unable to connect to my bluetooth device and receive the above message in my ConnectionStateUpdate.

Additionally, my app crashes with the following:
```
flutter_reactive_ble/PluginController.swift:153: Assertion failed
* thread #1, queue = 'com.apple.main-thread', stop reason = Assertion failed
    frame #0: 0x0000000191969060 libswiftCore.dylib`_swift_runtime_on_report
libswiftCore.dylib`_swift_runtime_on_report:
->  0x191969060 <+0>: ret
libswiftCore.dylib`_swift_reportToDebugger:
    0x191969064 <+0>: b      0x191969060               ; _swift_runtime_on_report
libswiftCore.dylib`_swift_shouldReportFatalErrorsToDebugger:
    0x191969068 <+0>: adrp   x8, 323753
    0x19196906c <+4>: ldrb   w0, [x8, #0x611]
```

I received the same connection error on Android sometimes but the crashing appears unique to iOS

JKornberg, 2021-07-06:
Connecting with connectToDevice instead of connectToAdvertisingDevice seems to resolve this

closed 2021-07-06 by JKornberg

remonh87, 2021-07-06:
@JKornberg  haopy it works one small remark: I do not know if you still use connectToAdvertisingDevice on Android since this is highly recommended on Android. Reason is that on Android if you try to connect to a device that is not reachable anymore it can cause the whole BLE stack to hang.

JKornberg, 2021-07-06:
@remonh87 interesting I will have to test more thank. My other issue however is making me doubt whether I can use this package or not https://github.com/PhilipsHue/flutter_reactive_ble/issues/309