PhilipsHue/flutter_reactive_ble #309: PlatformException The attribute could not be found when writing to BLE

JKornberg, 2021-07-06:
**Describe the bug**
My app was working in Android

**To Reproduce**
Steps to reproduce the behavior:
1. Connect using connectToDevice while manually inputting service and characteristic
2. Get qualified characteristic as follows:
```
final qualChar = QualifiedCharacteristic(
        serviceId: service,
        characteristicId: characteristic,
        deviceId: deviceId);
```
Where serviceId and CharacteristicId are gotten through Uuid.parse("<uuidstring>")

3. WriteCharacteristicWithoutResponse with values [36, 106, 48, 49] 
4. Crash with error

**Expected behavior**
It should write without crashing, it worked on Android but is not working on iOS Note: on android i was connecting with ConnectToAdvertisingDevice instead but this was causing issues on iOS. Additionally I used a bluetooth terminal app to ensure the characteristic and service ID's were correct.

**Smartphone / tablet**
 - Device: iPhoneX
 - OS: iOs 14.6
 - Package version: 3.1.1+1

**Peripheral device**
- Vendor, model: Custom
- Does it run a custom firmware / software:yes

**Additional context**

Another issue I'm having is the app crashing when the device goes to sleep, this did not happen on android. How do I deal with it?

remonh87, 2021-07-06:
Did you try the to write the characteristic using our bluetooth example app?  Also keep in mind that you need to discover the services before writing to the peripheral. Android does an auto discovery but on iOS you need to explictly run `discoverServices` if I am correct.

werediver, 2021-07-06:
Not exactly, @remonh87. On iOS a service discovery is done on connection, and there are two modes:

- discovering all services and characteristics present on a device (not recommended)
- discovering only what the app is going to actually use (recommended)

Most likely the issue here is that the services-and-characteristics to discover are not specified correctly. Try discovering everything.

remonh87, 2021-07-06:
Good one thanks for clarifying at least I knew sth was different 😊

JKornberg, 2021-07-06:
> Not exactly, @remonh87. On iOS a service discovery is done on connection, and there are two modes:
>
> * discovering all services and characteristics present on a device (not recommended)
> * discovering [...]

First off, here is a screenshot from the bluetooth terminal app so you can see the service and characteristic:

[image]

Here are my relevant functions.
For connecting-
```
 connectToDevice(DiscoveredDevice device) async {
    Uuid service = Uuid.parse("6E400001-B5A3-F393-E0A9-E50E24DCCA9E");
    Uuid characteristic = Uuid.parse("6E400002-B5A3-F393-E0A9-E50E24DCCA9E");
    try {
      await _connectionStateSubscription.cancel();
    } catch (error) {
      print(error);
    }
    _connectionStateSubscription = ble
        .connectToDevice(
      id: device.id,
      servicesWithCharacteristicsToDiscover: {
        service: [characteristic]
      },
      connectionTimeout: const Duration(seconds: 2),
    )
        .listen((connectionState) {
      print("New connection state: $connectionState");
      if (connectionState.connectionState == DeviceConnectionState.connected) {
        _device.sink.add(device);
        _connectionState.add(connectionState);
        writeResult(label: "Connected", value: "Connected to device");
      } else {
        _device.sink.add(null);
        _connectionState.add(null);
        writeResult(label: "Disconnected", value: "Lost connection to device");
      }
    }, onError: (dynamic error) {
      print("ERROR: " + error.toString());
      _connectionState.add(null);
    });
  }
```

For writing and listening:
```
  listenForJump({
    required String deviceId,
    required Protocol protocol,
    required int type,
    bool save = false,
    String? rosterId,
    String? athleteId,
  }) async {
    Uuid service = Uuid.parse("6E400001-B5A3-F393-E0A9-E50E24DCCA9E");
    Uuid characteristic = Uuid.parse("6E400002-B5A3-F393-E0A9-E50E24DCCA9E");
    List<int> airTimes = [];
    List<int> groundTimes = [];
    print(deviceId);
    print(service);
    print(characteristic);
    final qualChar = QualifiedCharacteristic(
        serviceId: service,
        characteristicId: characteristic,
        deviceId: deviceId);
    try {
      _bleReadWriteSubscription.cancel(); //close prior listen

    } catch (error) {
      print("Not subscribed yet");
    }
    writeResult(label: "Jump!", value: "Ready for Jump");
    _bleReadWriteSubscription = ble.subscribeToCharacteristic(qualChar).listen(
        (data) {
          print(data);
          String response = BluetoothUtil.decodeHex(data);
          List? r = BluetoothUtil.processResponse(response, protocol.code!);
          if (r != null) {
            //received termination

            if (airTimes.length == r[0]) {
              airTimes.add(r[1]['airTime']);
              groundTimes.add(r[1]['groundTime']);
            }
            if (r[0] == -1) {
              print("Received final jump for protocol!");
              print("Airtimes: $airTimes");
              print("Groundtimes: $groundTimes");
              List<List<dynamic>> res =
                  protocol.handleResults(airTimes, groundTimes);
              _result.add(res);
              cancelJump();
            }
          }
          print(response);
        },
        onError: (dynamic error) {
          print(error);
        },
        cancelOnError: true,
        onDone: () {
          print("DONE LISTENING");
        });
    String code = protocol.code!;
    List<int> readyData = BluetoothUtil.encodeString(code);
    ble
        .writeCharacteristicWithoutResponse(qualChar, value: readyData)
        .onError((error, stackTrace) => print(stackTrace));
    _listening.add(true);
  }
```

werediver, 2021-07-06:
I believe our `scanForDevices` method reports every advertisement received. It means that you should take care of the duplicate events for the same device, which doesn't seem to be implemented in the provided code (but maybe it is implemented elsewhere in your project).

@remonh87 Could you confirm whether our `scanForDevices` reports every advertisement received (_without_ deduplication)?

JKornberg, 2021-07-06:
> I believe our `scanForDevices` method reports every advertisement received. It means that you should take care of the duplicate events for the same device, which doesn't seem to be implemented in [...]

@werediver  The way I currently have it is the discoveredDevices are all added to a list on the screen and the user can click on one of them. Whichever discoveredDevice they select, that ID gets passed into the ConnectToDevice function

JKornberg, 2021-07-06:
Could there be an issue because my bluetooth device does not advertise any services?

werediver, 2021-07-06:
Hm, have you searched the web for that error message? It gives curious results.

For instance, check this: https://github.com/xabre/xamarin-bluetooth-le/issues/191#issuecomment-295389775

In short, your peripheral may not comply fully to the BLE spec.

werediver, 2021-07-06:
> Could there be an issue because my bluetooth device does not advertise any services?

That shouldn't be an issue as long as you're able to discover (scan for) your device and the services are actually discoverable on the device.

JKornberg, 2021-07-06:
> Hm, have you searched the web for that error message? It gives curious results.
> 
> For instance, check this: [xabre/xamarin-bluetooth-le#191 (comment)](https://github.com/xabre/xamarin-bluetooth-le/issues/191#issuecomment-295389775)
> 
> In short, your peripheral may not comply fully to the BLE spec.

I think this is likely the issue.

JKornberg, 2021-07-06:
I followed this thread: https://github.com/PhilipsHue/flutter_reactive_ble/issues/188
and it seems the compatibility fix is only for android? I actually know for a fact my BLE device does not conform to the correct CCCD. This package is working on Android for me but my troubles are in iOS

werediver, 2021-07-06:
If you have control over your peripheral (firmware), consider modifying it to implement the required descriptors.

Unfortunately, I doubt there is a workaround for iOS, regardless of the Flutter plug-in adapter (but I haven't researched the question).

JKornberg, 2021-07-06:
> If you have control over your peripheral (firmware), consider modifying it to implement the required descriptors.
> 
> Unfortunately, I doubt there is a workaround for iOS, regardless of the Flutter plug-in adapter (but I haven't researched the question).

The funny thing is, I was originally using Flutter Blue and it was working totally fine on iOS, but on Android I was getting an error regarding the CCCD. However now on this package it is vice versa where Android is functional and iOS is not

remonh87, 2021-07-24:
closed for now due to inactivity feel free to reopen it again and provide us with additional details so we can help you out.

closed 2021-07-24 by remonh87