PhilipsHue/flutter_reactive_ble #310: App crashing when is installed as apk release

josezainho, 2021-07-08:
**Describe the bug**
I have a fully functional app in debug mode.To generate the apk, I'm following the steps on the official flutter website.
Once I install the apk on the phone, the app runs, but when I click on the "scan" button, the app crashes with the following error:

java.lang.IllegalAccessError: Illegal class access: 'd.f.a.k0.A.d' attempting to access 'f.a.C.e.e.m' (declaration of 'd.f.a.k0.A.d' appears in /data/app/com.example.flutter_galan_bike-f3y4EPZj7fmFTugeMbS_2g==/base.apk)
	at d.f.a.k0.A.d.a(Unknown Source:16)
	at f.a.C.e.e.n.H(Unknown Source:10)
	at f.a.i.e(:2)
	at f.a.C.e.e.F1.H(Unknown Source:9)
	at f.a.i.e(:2)
	at f.a.C.e.e.o0.H(Unknown Source:9)
	at f.a.i.e(:2)
	at f.a.C.e.e.F.H(Unknown Source:17)
	at f.a.i.e(:2)
	at f.a.C.e.e.T.k(:6)
	at f.a.C.e.e.T.d(:2)
	at f.a.C.e.e.Z.H(:3)
	at f.a.i.e(:2)
	at f.a.C.e.e.U.H(Unknown Source:28)
	at f.a.i.e(:2)
	at f.a.C.e.e.r.H(:2)
	at f.a.i.e(:2)
	at f.a.C.e.e.o0.H(Unknown Source:9)
	at f.a.i.e(:2)
	at f.a.C.e.e.u0.H(Unknown Source:27)
	at f.a.i.e(:2)
	at f.a.i.G(:2)
	at com.signify.hue.flutterreactiveble.channelhandlers.ScanDevicesHandler.startDeviceScan(:1)
	at com.signify.hue.flutterreactiveble.channelhandlers.ScanDevicesHandler.onListen(Unknown Source:4)
	at io.flutter.plugin.common.EventChannel$IncomingStreamRequestHandler.onListen(Unknown Source:49)
	at io.flutter.plugin.common.EventChannel$IncomingStreamRequestHandler.onMessage(Unknown Source:22)
	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(Unknown Source:57)
	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(Unknown Source:4)
	at android.os.MessageQueue.nativePollOnce(Native Method)
	at android.os.MessageQueue.next(MessageQueue.java:336)
	at android.os.Looper.loop(Looper.java:181)
	at android.app.ActivityThread.main(ActivityThread.java:7561)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:539)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:995)

**Expected behavior**
When the app opens and I press "scan" (in debug mode) it lists the BLE devices to which I can connect, subscribe to their features, read/write...

**Additional context**

Here is an execution of flutter doctor -v
PS C:\Users\Jose\Desktop\reactive_ble_device> flutter doctor -v
[√] Flutter (Channel stable, 2.2.3, on Microsoft Windows [VersiÃ³n 10.0.19042.1052], locale es-ES)
    • Flutter version 2.2.3 at C:\Users\Jose\Desktop\Proyecto\flutter
    • Framework revision f4abaa0735 (7 days ago), 2021-07-01 12:46:11 -0700
    • Engine revision 241c87ad80
    • Dart version 2.13.4

[√] Android toolchain - develop for Android devices (Android SDK version 30.0.3)
    • Android SDK at C:\Users\Jose\AppData\Local\Android\sdk
    • Platform android-30, build-tools 30.0.3
    • Java binary at: C:\Program Files\Android\Android Studio\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b01)
    • All Android licenses accepted.

[√] Chrome - develop for the web
    • Chrome at C:\Program Files (x86)\Google\Chrome\Application\chrome.exe

[√] Android Studio (version 4.1.0)
    • Android Studio at C:\Program Files\Android\Android Studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b01)

[√] VS Code (version 1.57.1)
    • VS Code at C:\Users\Jose\AppData\Local\Programs\Microsoft VS Code
    • Flutter extension version 3.24.0

[√] Connected device (3 available)
    • M2003J15SC (mobile) • 1008c34c0410 • android-arm64  • Android 10 (API 29)
    • Chrome (web)        • chrome       • web-javascript • Google Chrome 91.0.4472.124
    • Edge (web)          • edge         • web-javascript • Microsoft Edge 91.0.864.64

• No issues found!

**Proguard.rules:**
#Flutter Wrapper
-keep class io.flutter.app.** { *; }
-keep class io.flutter.plugin.**  { *; }
-keep class io.flutter.util.**  { *; }
-keep class io.flutter.view.**  { *; }
-keep class io.flutter.**  { *; }
-keep class io.flutter.plugins.**  { *; }
-keep class com.signify.hue.** { *; }
-keep class androidx.appcompat.**{ *;}

**build.gradle file fragment**
buildTypes {
        release {
            signingConfig signingConfigs.release
             minifyEnabled true
             useProguard true

             proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'

        }
    } 
**Versions**
pubspec.yalm:
flutter_reactive_ble: ^3.1.0
build.gradle:
minSdkVersion 21
targetSdkVersion 30

remonh87, 2021-07-12:
This looks like a proguard issue indeed. Maybe try adding

can you change signify entry to: 
```
-keep class com.signify.hue.** { *; }

```

josezainho, 2021-07-13:
Hi, thank you for your response.
 
Sorry, it looks like a format error from the last time I wrote here. This is my proguard file:

[image]

As you can see I copied propertly the entry from the library. 

Another idea? Thank you!

remonh87, 2021-07-15:
maybe it is protobuf try:

`-keep class com.google.protobuf:protobuf.** { *; }`

josezainho, 2021-07-16:
I run ./gradlew app:dependencies. There is not another dependency using protobuf, only reactive_ble. 
Adding -keep class com.google.protobuf.** { *; } in the proguard file did not work.  Same problem persists.

remonh87, 2021-07-19:
I think it is something related to your dependencies and proguard which is a tough problem. What I would do is create a sample app and add dependencies one by one and try to figure out where it crashes. Unfortunately there is not a lot that I can fix from our side at the moment.

closed 2021-07-19 by remonh87