PhilipsHue/flutter_reactive_ble #312: Incorrect connection state change when failed writeCharacteristicWithoutResponse()

atauf, 2021-07-19:
**Describe the bug**
I am writing to a characteristic using writeCharacteristicWithoutResponse(), but it fails (I believe this is due to the device itself, not this package). However, when this happens, about 30 seconds or so later, the connection state changes to be disconnected, even though it is actually still connected to the device. (I've verified this by looking at the device through a serial connection).

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to device
2. Monitor device connection through serial connection or other
2. write to device using writeCharacteristicWithoutResponse() (this must be a device that doesn't allow the designated characteristic to be written, or has an issue doing so at a minumum. Thus, providing a failed attempt).
3. Wait about 30 seconds and notice that the device connection state read by the app says that it is now 'disconnected'
4. Verify that the device is actually still connected

**Expected behavior**
What should happen when the failed 'writeCharacteristicWithoutResponse()' goes through is that no change in device connection state takes place (I think).

**Smartphone / tablet**
 - Device: Phone - Pixel 4 XL
 - OS: Android 11
 - Package version: 3.0.0

**Additional Comments**
I don't know if this is just part of the way a failed write works or not. I'm thinking not, but if it is, feel free to remove the issue.

remonh87, 2021-07-20:
Can you help me out by sharing some example code and maybe provide some meaningful logs?  You are right that the device should not be getting disconnected however there may be some reasons for this outside this library (BLE stack on the device get's into a weird state and cannot recover, the peripheral crashes for some reason).  The logs and exampel code would help me with identifying the issue.

atauf, 2021-07-20:
@remonh87 What you're mentioning about the issue with the BLE stack may actually be what's creating the issue. I'm trying to figure this out at the moment, and will update you as soon as I know. For now, this might not be worth your time to work on. I will hopefully have an update in a day or two.

atauf, 2021-07-29:
@remonh87 Sorry for the delay. It looks like there was an issue in the BLE stack. It seems to be taken care of, so I think it's safe to close this issue.

remonh87, 2021-07-29:
Good one I enabled discussions for small questions like these. Btw different advertising rates is part of BLE protocol. I think this article will help you: https://www.novelbits.io/bluetooth-low-energy-advertisements-part-1/

closed 2021-07-29 by remonh87

atauf, 2021-07-30:
Thanks @remonh87 . I wrote a couple discussion Q&A's. If you get a chance to check I'd really appreciate any input.