PhilipsHue/flutter_reactive_ble #313: Gradle related build error with flutter_reactive_ble 3.1.1+1 & Flutter 2.2.3

archanpaul, 2021-07-20:
**Describe the bug**
Build failure with flutter_reactive_ble: ^3.1.1+1
```
Flutter version used : 
Flutter 2.2.3 • channel stable • https://github.com/flutter/flutter.git
Framework • revision f4abaa0735 (3 weeks ago) • 2021-07-01 12:46:11 -0700
Engine • revision 241c87ad80
Tools • Dart 2.13.4
```

Errors reported : 
```
[ +478 ms] FAILURE: Build failed with an exception.
[   +1 ms] * What went wrong:
[        ] A problem occurred configuring project ':flutter_reactive_ble'.
[        ] > class com.android.build.gradle.internal.api.TestVariantImpl_Decorated cannot be cast to class com.android.build.gradle.internal.core.InternalBaseVariant (com.android.build.gradle.internal.api.TestVariantImpl_Decorated is in unnamed module of loader
org.gradle.internal.classloader.VisitableURLClassLoader @aca112e; com.android.build.gradle.internal.core.InternalBaseVariant is in unnamed module of loader org.gradle.internal.classloader.VisitableURLClassLoader @273919cd)
```

**To Reproduce**
Steps to reproduce the behavior:
1. flutter clean
2. flutter pub get
3. fluter run -v
4. 
**Expected behavior**
Should get compiled without any issues. This build failure is seen after Flutter (stable) is upgraded to the latest.

**Smartphone / tablet**
 - Device: Samsung Galaxy N9
 - OS: Android 10
 - Package version: 3.1.1+1

remonh87, 2021-07-20:
Make sure you upgrade the grade wrapper to version 6.8.1 . This issue is related to this.

soumyapal-arputer, 2021-07-20:
Updated gradle wrapper to 6.5 and gradle build tools to 4.1.3. This resolves the build issue.

closed 2021-07-21 by archanpaul