PhilipsHue/flutter_reactive_ble #314: How to work with workmanager plugin?

yo80106, 2021-07-23:
Recently, I am trying to connect my Flutter app with an Arduino device through BLE. Everything had been working perfectly in the foreground until I tried to make the connection in the background. I use the [workmanager plugin](https://pub.dev/packages/workmanager) to handle the periodic task in the background. I put the BLE scanning and connection in the callback dispatcher. However, it seems nothing has happened with the BLE part. Can anyone help me with this issue? I realized that background processing in the Flutter may need to write some native code but I am trying to use the plugin to do the task.

remonh87, 2021-07-24:
Hi I have no knowledge about the workmanager plugin maybe the  maintainers of that plugin can help you out.  There are no restrictions in this library that would prevent BLE to work in the background. Can you check if the methodcall is executed when in background? 

 As alternative I found this article of background processing in Flutter quite helpful when I needed background processing: https://medium.com/flutter/executing-dart-in-the-background-with-flutter-plugins-and-geofencing-2b3e40a1a124

yo80106, 2021-07-24:
Thank you for your suggestion. I will go for study further in background processing. After I initialized FlutterReactiveBle in the callback dispatcher, I can get the updated BLE status (first line of the code `print(_ble.status);`). However, the rest of the BLE code seems not working because none of any print statements is executed within the block.
```
print(_ble.status);
_subscription = _ble.scanForDevices(
    withServices: [serviceID],
    scanMode: ScanMode.lowLatency,
    requireLocationServicesEnabled: false).listen((device) {
      print(device.name);
  if (device.name == 'PersonDetection') {
    print('PersonDetection found!');
    _connection = _ble
        .connectToDevice(
      id: device.id,
      connectionTimeout: const Duration(seconds: 5),
    )
        .listen((connectionState) async {
      // Handle connection state updates
      print(device.name);
      print(device.id);
      print('connection state:');
      print(connectionState.connectionState);
      if (connectionState.connectionState ==
          DeviceConnectionState.connected) {
        final characteristic = QualifiedCharacteristic(
            serviceId: serviceID,
            characteristicId: characteristicsID,
            deviceId: device.id);
        final response = await _ble.readCharacteristic(characteristic);
        print(response);
        _disconnect();
        print('disconnected');
      }
    }, onError: (dynamic error) {
      // Handle a possible error
      print(error.toString());
    });
  }
}, onError: (error) {
  print('error!');
  print(error.toString());
});
```

remonh87, 2021-07-25:
I do not know the exact internals of the plugin but I still think you need to tie your to a longer running process because you need to await the results.  What would be good is to check the android logcat and filter it on ble. If the scanning is called you normally would see something related to scan in the logs.

yo80106, 2021-07-25:
Thank you for your suggestion. I will study deeper to use the workmanager. I think the issue is nothing to do with flutter_ble_reactive. I will close the issue.

closed 2021-07-25 by yo80106

yo80106, 2023-06-28:
@sifourquier , I gave up to use this plugin for background connection and used the Kotlin.