PhilipsHue/flutter_reactive_ble #316: write long byte arrays

HelloBytes, 2021-07-27:
Hello,

I need to write a byte array of 950 bytes. In ios it works, in android it doesn't.

Some errors I found in Logcat:

E/bt_stack: [ERROR:gatt_cl.cc(569)] value resp op_code = ATT_RSP_PREPARE_WRITE len = 250

Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "GATT exception from MAC='XX:XX:XX:XX:XX:XX', status 9 (GATT_PREPARE_Q_FULL), type BleGattOperation{description='CHARACTERISTIC_WRITE'}. (Look up status 0x09 here https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h)")

remonh87, 2021-07-27:
I think this has to do with the phone vs peripheral combination. Both need to negotiate an MTU together and if you try to write blocs that exceed that MTU you will get this error. What I would suggest first is:
- Negotiate an MTU 
- based on the number provided make sure that you chunk your bytearray accordingly so it does not exceed

I also ran into this [useful source](https://e2e.ti.com/support/wireless-connectivity/bluetooth-group/bluetooth/f/bluetooth-forum/464746/can-t-send-more-than-90-bytes-ble-android-cc2640). Hope it helps

HelloBytes, 2021-07-27:
Thank you for your fast reply

For example, I can negotiate 251 MTU on my device
The byte array is 950. 

I can write a single 950 with writeCharacteristicWithResponse (does work with ios)

Or do I have to chunk the 950 in 4 parts (~236 bytes) and send them with writeCharacteristicWithResponse.

Both tried without success. 

In the RxAndroidBle library I see a method called: createNewLongWriteBuilder. 

Is writeCharacteristicWithResponse the right way to send large byte arrays? The peripheral expects a 950 bytes array, not chunks of data.

remonh87, 2021-07-28:
the createNewLongWriteBuilder does in fact the same as what I suggested. Thing is that if the phone does not support sending 950 bytes you need to adapt the peripheral so it accepts chunks of various length.  There is little this library can do to help you out.

closed 2021-07-28 by remonh87

HelloBytes, 2021-07-28:
Thank you Remon,

I asked the manufacturer to adjust the characteristic to accept smaller pieces, there is no other way.