PhilipsHue/flutter_reactive_ble #317: Document RSSI range

petoknm, 2021-07-28:
What is the range of `rssi` values I should be expecting? I can't find this information in the documentation.

Thanks :)

petoknm, 2021-07-28:
[maybe this?](https://developer.android.com/reference/android/net/wifi/WifiManager#calculateSignalLevel(int))

remonh87, 2021-07-28:
RSSI is very volatile and complex and really depends on the strength of the radio from the phone and peripheral + interference (other BLE devices, WiFI etc.) . I think what google states is not a bat range to start with but doesn't mean you can get below or above it.

remonh87, 2021-07-28:
And there is a difference between platforms. In general we found out that iOS devices tend to have more beneficial RSSI values compared to Android devices for our lights.

closed 2021-07-28 by remonh87