PhilipsHue/flutter_reactive_ble #322: Disconnect already connected Device after App restart

bysonstudio, 2021-08-01:
in advance, thanks for your great job !

i have a question...

```
■ Scenario 0
1. connect to some device
2. disconnect above device
-----> doing well
```

```
■ Scenario 1
1. connect to some device
2. terminate app
3. restart app
-----> we can find connected device by
-----> listening connectedDeviceStream
4. how can disconnect above device's connection?
```

here is the problem or my ignorance

```
How can i disconnect device what found by connectedDeviceStream
not from manually connectToDevice nor connectToAdvertisingDevice ?

i already know canceling stream from connectToDevice
or connectToAdvertisingDevice take disconnection

but, connected device from connectedDeviceStream have no stream
so i can't disconenct those devices..... T_T
```

```
■ Trying
after Scenario 1, if i found already connected device
then make connection and cancel...

but can't get disconnected callback by ConnectionStateUpdate
(both stream from connection and FlutterReactiveBle.instants's connectedDeviceStream)

also can't scan that device...
```

Can you do me a favor ?

remonh87, 2021-08-02:
That is an interesting scenario indeed and not very uncommon (especially on Android).  I would just try to connect to the device anyway, since on the OS BLE layer  there is already a connection you will get super quick response.  There are cases even that despite we send a disconnect command, the OS decide to keep the connection alive so other apps can use it. 

However it is weird in my opinion that you will get a connected device update back from our Plugin after terminating it. Do you use it in background mode?

bysonstudio, 2021-08-03:
by your plugin

1. app terminate with force closing, release all connection
2. just close app using back button, keep connection

isn't it right?

remonh87, 2021-08-14:
this is right but I think you should be able to reconnect to the device if I am correct.

jayjhaveri, 2021-09-13:
I am facing this issue, too. Our application is if the device is connected to a mobile then BLE's Bluetooth locked (off), so if the user clears the ram and opens the app again then we can't see the BLE while scanning, we have to switch off and on the Bluetooth to break the connection. Is there any way to detect all connected devices?

jahnli, 2021-09-13:
@remonh87  In fact, if it was a thermal reload, it would not disconnect the device and I would not be able to scan files to the device

jahnli, 2021-09-24:
For example and if you use the deviceID as key then it is quite simple to disconnect specific devices

remonh87, 2021-09-27:
I understand the issue we can provide a list of connected devices and a disconnect command. However using plugins with hot restart can yield to unexpected behaviour. Even firebase logs warnings when doing a hot restart so not sure it will fully solve the issue. 

The enhancement though is a good one to have for this library.

jahnli, 2021-09-28:
> 我了解我们可以提供已连接设备列表和断开连接命令的问题。然而，使用带热重启的插件可能会导致意外行为。即使 firebase 在进行热重启时也会记录警告，因此不确定它是否会完全解决问题。
> 
> 不过，增强对于这个库来说是一个很好的增强。

jahnli, 2021-09-28:
@remonh87  When will this feature be available now, I see a lot of people need it

remonh87, 2021-10-31:
implemented android functionality and coming week I will work on iOS

xuyongfh, 2021-11-11:
hope for this enhancement , my BLE server is a  global status . Users will not actively disconnect Bluetooth. A temporary solution is to use "exit (0)" to exit App, then BLE disconnect immediately， but it's not recommended as it terminates the dart VM process immediately and user may think that the app just got crashed

rohitsangwan01, 2021-12-05:
@remonh87  any progress for this feature , my app really depends on this one

xuyongfh, 2021-12-05:
My App also waiting for this  , check this update everyday

remonh87, 2021-12-08:
I would raise caution against this workaround. BLE on Android is quite challenging and it is important that you properly orchestrate the operations because if they are out of sync you run into weird issues (e.g. the famous GATT 133 error can be one ). We spend a lot of time making this orchestration good and I am sure also FlutterBLue did the same but implementing two different libraries in one app can cause nasty side effects. Personally it   is something I wouldn't do.

About disconnect: I did some study on the Android API and what it does is first create a connection and then disconnect it . Also read this very interesting post on Nordic Semi conductors forum: https://devzone.nordicsemi.com/f/nordic-q-a/38422/android-disconnect-and-close-do-not-disconnect . So even if you tell Android to disconnect there is no guarantee this will actually happen.  I did some tests on my Android test implementation and I observed the same.

Another interesting challenge is the `connectedDevicesi` api  if a device is reported to be connected  it does not mean your app has connected it in the first place. It can be another app is using this connection. https://developer.android.com/reference/android/bluetooth/BluetoothManager#getConnectedDevices(int)

To summarise there is no bullet proof solution on Android and this makes me hesitant to implement it. It is quite easy to misunderstand the use of the operations and it masks the actual problem and that is that a peripheral should accept a connection request or be able to advertise when it is connected.

rohitsangwan01, 2021-12-08:
@remonh87 , yeah sure there are lots of issue using multiple ble libraries , and am facing this 133 again and again in Few Device , not in mine (Stock Andriod 11) , but client is reporting this (Samsun Android V9 ) , is there any good fix for this , i already read issue related to this here , but none of them seems to work , tried to toggle (Auto / Timeout), added Delay of 2 sec. between scan Stream Close and connection Stream call , 
And Then again , sometimes Device remains connected even if we close the Stream , thats really annoying in Android

remonh87, 2021-12-15:
I had a discussion with the other maintainers and based on the above findings we decided not to implement this feature. We think that to know the previously connected devices you can implement specific logic in your app (e.g. if needed write the device id to shared preferences). Then after restart you should be able to connect the device.

However a better option is to improve the firmware on the peripheral  e.g. making sure that the device can handle multiple connection requests, or is able to be discovered when having a connection. For the Hue App we never felt a need to implement this feature because of this.

closed 2021-12-15 by remonh87

cbreezier, 2022-04-23:
Hi @remonh87 , I've read through this thread a few times now, to make sure I'm understanding all the context.

> For the Hue App we never felt a need to implement this feature because of this.

While I understand where you're coming from, I'd love for this package to be able to serve the needs of many, not just for the Hue App.

> However a better option is to improve the firmware on the peripheral

Sometimes this isn't an option.

> Another interesting challenge is the connectedDevicesi api if a device is reported to be connected it does not mean your app has connected it in the first place. It can be another app is using this connection.

This is a strength, in my opinion. For better or for worse, the bluetooth stack is shared across multiple applications. Sometimes it is incredibly useful to be able to disconnect a device which was connected to from another application. The way the `flutter_reactive_ble` package works today, it can only know about devices which it has initiated connections to. There is an implicit assumption that the "current app" is the only one initiating BLE connections. Unfortunately, when this assumption is broken, we end up with unsolvable problems without any recourse.

> We think that to know the previously connected devices you can implement specific logic in your app (e.g. if needed write the device id to shared preferences). Then after restart you should be able to connect the device.

This still assumes that other apps don't exist, or that the previous app shutdown happened gracefully enough to store information about previously connected devices.

The reality is that the Android API can tell us, somewhat reliably, whether a particular device is currently connected to or not. But `flutter_reactive_ble` doesn't make that information available to us.

I'd like to appeal to the maintainers to reconsider. Perhaps these "force disconnect" and "list all connected" apis can be marked as "unsafe", or come with extra documentation explaining their limitations.

jahnli, 2022-04-24:
Yes, we have the same needs and would like to appeal to the maintainer to reconsider

figureai, 2022-08-02:
Is this issue fixed?

thumptech, 2022-08-08:
Please just implement it.  It's a total ballache having connection leaks on hot restarts.

phimary, 2022-09-14:
Are we already have possibility to get connected devices after restarting an app?

This library is great, but still needs that functionality. My problem is next on iOS device: start app -> connect to BT device -> pair, after pairing when I restart the app I can't detect this BT point in the app, because there is already connection in iOS OS.

akospwc, 2022-10-25:
I second the request. We are stuck when the app starts with an existing active connection to the peripheral. Cannot disconnect because there is no connection stream.

jonashoelzel, 2023-10-12:
Are there any updates on a solution for hot restart?

a1573595, 2023-10-25:
Please implement this feature

sreetejadusi, 2025-04-18:
I too have been facing this issue. I currently use flutter_blue_elves package, and when I abruptly close the app, from the recents too- and reopen it. The device won't reconnect. When I hot-restart, the device won't disconnect. 

Is there any way I can clear the bluetooth connections, to make it afresh everytime the app is opened? or,
Is there a way I can get existing connections via bluetooth, so I can terminate them?