PhilipsHue/flutter_reactive_ble #327: Unable to disconnect from all connected devices

atauf, 2021-08-04:
**Describe the bug**
I am unable to disconnect from multiple devices. I successfully connect to them using a for loop, but when I try to use a for loop OR a single button press to disconnect from more than one device, it doesn't work. I am only ever able to disconnect from the first device that I request to disconnect from. After that, the disconnect request doesn't fully execute, and no exception/error is thrown.

**To Reproduce**
Steps to reproduce the behavior are explained in the Discussion #326 

**Expected behavior**
- Given a list of device ids, (of which pertain to currently connected devices), I want to be able to successfully disconnect from them all.

- Also, if a disconnect does not fully complete, I think there should be an exception or something thrown to make the developer away. This doesn't happen (but the connectionState says Disconnected (even though it's not).

**Smartphone / tablet**
 - Device: Pixel 4 XL
 - OS: Android 11
 - Package version: 3.1.1+1

**Peripheral device**
- Does it run a custom firmware / software: YES

**Additional context**
The only other piece of info I have is that all devices successfully connect when I close the app, but I don't want to have to do that.

Also, I'm not sure if this has anything to do with it, (but I only mention it because I also just came across this problem and implemented a fix), but when trying to disconnect, if there are readCharacteristic() calls on the page, it seems to be that the device auto connects again to read. The fix I did was that I wrapped all readCharacteristic() calls in a condition that checks if connectionState was connected or not before it tried to read.

atauf, 2021-08-05:
UPDATE:
It looks like when I step into the debugger a bit more I see an error get thrown, (but not mentioned in the console).
It is:
```
Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "failed to connect Disc..."
messge: failed to connect Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 8 (GATT_INSUF_AUTHORIZATION or GATT_CONN_TIMEOUT)
```

atauf, 2021-08-05:
UPDATE 2:

As I debug more, it looks like the command for disconnecting one deviceId gets sent, but the next deviceId in my deviceIdList is actually the one that performs the **cancelOpen()**

So, my console log looks like this after the first time I press the button (this is when the button functionality is to just disconnect index 0, and then the next time the button is press it will disconnect using index 1, and so on):

```
// Note: Device Id List looks like this -> ["##:##:##:##:##:01", "##:##:##:##:##:02"]

Disconnected from ##:##:##:##:##:01  // print statement added to the .disconnect(String deviceId) function
D/BluetoothGatt( 6644): cancelOpen() - device: ##:##:##:##:##:02
D/BluetoothGatt( 6644): onClientConnectionState() - status=0 clientIf=11 device=##:##:##:##:##:02
D/BluetoothGatt( 6644): close()
```

This really seems like a bug to me, but I just don't know if others are having this issue or not. I don't know why it changes up the deviceId that it was supposed to disconnect from.

remonh87, 2021-08-07:
What can help is that you put a small delay between disconnecting each device (I would recommend around 500ms).  Android does not like ble operations fired in a quick succession.

atauf, 2021-08-09:
@remonh87 That's actually what I did and it ended up working. Thanks.

closed 2021-08-14 by remonh87